cpwood/Pico-Go #1: "No PyCom boards found on USB", but other REPL tools outside of VS Code work fine

markdall, 2021-01-29:
## What are the steps to reproduce this issue?
1. Launch VS Code.
2. Observe error message.

## What happens?

Searching for boards on serial devices...
No PyCom boards found on USB

## What were you expecting to happen?
The Raspberry Pico shows up as connected to COM3 (I verified that it still works, etc.)

## Any logs, error output, etc?
Searching for boards on serial devices...
No PyCom boards found on USB

## Any other comments?
Micropython is installed and functional, I can make the LED blink, etc. via for example Thonny. Telling Pico-Go to connect or disconnect doesn't seem to do anything, and I couldn't find anywhere in the settings to specify which COM port to connect to. 

## What versions of software are you using?

**Operating system:**
Windows 10 version 2004 21296.1010

**VSCode version:**
VS Code version 1.52.1

**Pico-Go version:**
Pico-Go version 1.0.1

**NPM Version:**
6.14.10

cpwood, 2021-01-29:
Thanks for the heads-up! I can see the problem and will fix it tonight.

Once VS Code has loaded, press Ctrl + Shift + G and in the array that has "MicroPython" in it, also add "Microsoft".

```
	"autoconnect_comport_manufacturers": [
		"MicroPython",
		"Microsoft"
	]
```

If you get the message:

```
Failed to connect (Error: Port is not open). Click here to try again.
```

Physically unplug your Pico and then plug it in again.

Will get this fixed pronto and will remove the reference to Pycom.

cpwood, 2021-01-29:
Version 1.0.2 fixes this where Pico-Go has never been installed before, however you will probably need to update your config manually as above if you'd already downloaded 1.0.1

closed 2021-01-29 by cpwood

markdall, 2021-01-29:
That worked, thank you so much!

trepidacious, 2021-01-30:
I seem to be getting `> Failed to connect (Error: Port is not open). Click here to try again.` repeatedly even if I disconnect and reconnect?

cpwood, 2021-01-30:
Sorry to hear that. Here's a few questions / things to try:

- Is it trying to connect to the correct COM port (Windows) or device (macOS or Linux)?
- Is there another piece of software running that might already be connected to the Pico (e.g. Thonny)?
- Are you able to connect from another piece of software, such as Thonny?
- Tried rebooting your computer?
- What OS are you running?
- Do you have multiple serial-based devices attached to your computer?

trepidacious, 2021-01-31:
Thanks - when I set out to try those out it connected immediately, my PC had been asleep overnight with the pico disconnected, other than that nothing changed except waking the PC up and connecting the pico.

> * Is it trying to connect to the correct COM port (Windows) or device (macOS or Linux)?

Looks correct - it's showing COM port 3

> * Is there another piece of software running that might already be connected to the Pico (e.g. Thonny)?

Not that I could see, I don't have any serial software installed, or anything else much running.

> * Are you able to connect from another piece of software, such as Thonny?

Not tried.

> * Tried rebooting your computer?

Not tried, but it did sleep for about 10 hours.

> * What OS are you running?

Windows 10. I've also tried on Linux (pop os) where it worked pretty much straight away, although it has been occasionally getting stuck until the PC is rebooted.

> * Do you have multiple serial-based devices attached to your computer?

No.

commit ea95d70e (2021-01-29): Fix #1 in commit 6cc22b2