cpwood/Pico-Go #11: VSCode 1.53.0 incompatibility

snympi, 2021-02-04:
## What are the steps to reproduce this issue?
1. Update VSCode to 1.53.0
2. Upon reload error message stating version incompatibility
3. Extension does not activate and functionality is not available

## What happens?
Error message is received saying: 'This version of Pico-Go is incompatible with VSCode 1.53.0. Check for an update to the extension. If one isn't available, raise a bug at https://github.com/cpwood/Pico-Go to get this fixed!'. Extension does not load.

## What were you expecting to happen?
Extension should activate with functionality available

## Any logs, error output, etc?

## Any other comments?

## Which versions of software are you using?

**Operating system: Windows 10, 19042.685

**VSCode version: 1.53.0

**Pico-Go version: 1.0.5

## Which hardware are you using?
Raspberry Pi Pico RP2040 board

cpwood, 2021-02-04:
Looking at this and will hopefully finish tomorrow (Friday 5 Feb). I'm seeing connectivity issues on a Mac after updating the components for the new version of VS Code; will check Windows tomorrow and see if it's likewise impacted.

RamblingGeekUK, 2021-02-04:
Just got this on windows, if it helps. 

"This version of Pico-Go is incompatible with VSCode 1.53.0. Check for an update to the extension. If one isn't available, raise a bug at https://github.com/cpwood/Pico-Go to get this fixed!"

cpwood, 2021-02-04:
If anybody wants to try an early version of 1.0.6, you can download it here:

https://cwood.blob.core.windows.net/temp/pico-go-1.0.6.vsix

Mainly curious to know two things:

1. Has the warning about Pico-Go being incompatible gone away?
2. Are you running into the same connection issue as me above? If so, which OS are you on?

Thanks in advance!

snympi, 2021-02-05:
I tried 1.0.6 and it loads the extension in VS Code 1.53.0 but it fails to connect to the board
[image]

Naohiro2g, 2021-02-05:
Hi, today, I've got the Pico-go v1.0.5 and it was working fine with VS Code on my Mac for a while. Actually I've been using pyMakr, so I had double menu at the bottom.
Eventually I noticed the VS Code update and got v1.53.0 then both were stop working 
with the incompatibility warning.

Then I tried the 1.06, the warning message were gone but Pico-go started to try local ip connection which I used for EPS32 before.  Ah, if auto_connect is false, it tries the modem port, right one but in vain.

```
Connecting to /dev/tty.usbmodem0000000000001...
> Connection timed out. Click the "Pico Disconnected" button to try again.
```

cpwood, 2021-02-05:
I *think* the problem is with the [pre-built "`serialport` stubs](https://github.com/serialport/node-serialport/releases/tag/%40serialport%2Fbindings%409.0.4) for Electron ABI 85. If I build them myself it all works fine; if I use the pre-built ones nothing connects (or at least that's the case on my Mac). Working on a solution now.

mkshrps, 2021-02-05:
Just ran the picogo.1.0.6.vsix file on VSode q.53.0. It now loads ok but hits same problem with connection timeout. running  on mac catalina 10.15.7

cpwood, 2021-02-05:
Thanks for the feedback, everyone. I have made some progress today but it's been slightly interrupted by real life. Will get a bit more done later and will update you all. Thanks for your patience.

deeteeakl, 2021-02-05:
Just found this after struggling with getting it working.  Tried with the new vsix (on my mac) and its working. Great stuff!

cpwood, 2021-02-06:
Look out for 1.0.7 very shortly via the Marketplace! We now rely on self-built `bindings.node` files for `@serialport`; this also means we can look to support arm64 (i.e. Raspberry Pi!) in a future release too, since pre-built bindings don't exist from the original developer.

closed 2021-02-06 by cpwood

snympi, 2021-02-06:
@cpwood I've just tested 1.0.7 and it work in VS Code 1.53.0 and VS Code Insider 1.54.0. Great job, thanks.

Naohiro2g, 2021-02-06:
Great! It's working on my Mac, too.

But it still need a fix on the case when auto_connect is false. Do we still need the option for the manual connection?

cpwood, 2021-02-06:
@Naohiro2g that's being tracked as #12 . Will get to it as soon as I can.

mkshrps, 2021-02-06:
just installed via marketplace on mac catalina all works so far Thanks :-)

studer, 2021-02-07:
I've still got the same error message :

> This version of Pico-Go is incompatible with VSCode 1.53.0-exploration. Check for an update to the extension. If one isn't available, raise a bug at https://github.com/cpwood/Pico-Go to get this fixed!

I've installed pico-go 1.0.7 from the marketplace.

Error appears either with VS Code Insider 1.54.0 or VS Code Exploration 1.53.0, both on MacBook Air with M1 without Rosetta 2.

cpwood, 2021-02-07:
@studer at the moment, only x64 is supported, so you'd need Rosetta. 

I'm actually working on arm64 support for *Linux* at this _very_ moment since it would also make sense to support the Raspberry Pi 4 for a Raspberry Pi Pico :-)

But I'd need the capability to build for Darwin arm64 using electron-rebuild too. Not sure if that's available yet. It'll be a while before I get to it, but it's on the list. In fact, I'll add it to the issues list.

andypiper, 2021-02-07:
Ah, looks like my #17 is related here - I'm seeing this error on RasPi 400 w/ 1.53.0

commit fcb73794 (2021-02-06): Fix #11 with commit 967148e