cpwood/Pico-Go #9: command 'pymakr.connect' not found

renanqts, 2021-02-04:
## What are the steps to reproduce this issue?
1. Go to the command palette
2. Type Pico-Go > Connect
3. The message above will show up

[image]

## What were you expecting to happen?
It works out of the box.

## Any logs, error output, etc?
No

## Any other comments?
No

## Which versions of software are you using?

**Operating system:**
Ubuntu 20.04

**VSCode version:**
1.53.0

**Pico-Go version:**
1.0.4

## Which hardware are you using?
RPi Pico

cpwood, 2021-02-04:
Confirming I can reproduce this after having upgraded to VSCode 1.53 from 1.52.

Exception upon startup is as follows:

```
Error: The module '/Users/chris/Source/Repos/Pico-Go/node_modules/@serialport/bindings/build/Release/bindings.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 83. This version of Node.js requires
NODE_MODULE_VERSION 85. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).
	at process.func [as dlopen] (electron/js2c/asar_bundle.js:5:1812)
	at Object.<anonymous> (internal/modules/cjs/loader.js:1250:18)
	at Object.func [as .node] (electron/js2c/asar_bundle.js:5:1812)
	at Module.load (internal/modules/cjs/loader.js:1039:32)
	at internal/modules/cjs/loader.js:932:14
	at Function.<anonymous> (electron/js2c/asar_bundle.js:5:12738)
	at Function.<anonymous> (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:104:27152)
	at Function.<anonymous> (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:104:23904)
	at Function._load (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:94:62159)
	at Module.require (internal/modules/cjs/loader.js:1079:19)
	at require (internal/modules/cjs/helpers.js:72:18)
	at bindings (/Users/chris/Source/Repos/Pico-Go/node_modules/bindings/bindings.js:112:48)
	at Object.<anonymous> (/Users/chris/Source/Repos/Pico-Go/node_modules/@serialport/bindings/lib/darwin.js:2:36)
	at Module._compile (internal/modules/cjs/loader.js:1199:30)
	at Object..js (internal/modules/cjs/loader.js:1220:10)
	at Module.load (internal/modules/cjs/loader.js:1039:32)
	at internal/modules/cjs/loader.js:932:14
	at Function.<anonymous> (electron/js2c/asar_bundle.js:5:12738)
	at Function.<anonymous> (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:104:27152)
	at Function.<anonymous> (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:104:23904)
	at Function._load (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:94:62159)
	at Module.require (internal/modules/cjs/loader.js:1079:19)
	at require (internal/modules/cjs/helpers.js:72:18)
	at Object.<anonymous> (/Users/chris/Source/Repos/Pico-Go/node_modules/@serialport/bindings/lib/index.js:10:22)
	at Module._compile (internal/modules/cjs/loader.js:1199:30)
	at Object..js (internal/modules/cjs/loader.js:1220:10)
	at Module.load (internal/modules/cjs/loader.js:1039:32)
	at internal/modules/cjs/loader.js:932:14
	at Function.<anonymous> (electron/js2c/asar_bundle.js:5:12738)
	at Function.<anonymous> (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:104:27152)
	at Function.<anonymous> (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:104:23904)
	at Function._load (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:94:62159)
	at Module.require (internal/modules/cjs/loader.js:1079:19)
	at require (internal/modules/cjs/helpers.js:72:18)
	at Object.<anonymous> (/Users/chris/Source/Repos/Pico-Go/node_modules/serialport/lib/index.js:2:17)
	at Module._compile (internal/modules/cjs/loader.js:1199:30)
	at Object..js (internal/modules/cjs/loader.js:1220:10)
	at Module.load (internal/modules/cjs/loader.js:1039:32)
	at internal/modules/cjs/loader.js:932:14
	at Function.<anonymous> (electron/js2c/asar_bundle.js:5:12738)
	at Function.<anonymous> (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:104:27152)
	at Function.<anonymous> (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:104:23904)
	at Function._load (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:94:62159)
	at Module.require (internal/modules/cjs/loader.js:1079:19)
	at require (internal/modules/cjs/helpers.js:72:18)
	at prepareSerialPort (/Users/chris/Source/Repos/Pico-Go/pymakr.js:154:9)
	at activate (/Users/chris/Source/Repos/Pico-Go/pymakr.js:7:5)
	at Function._callActivateOptional (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:90:13728)
	at Function._callActivate (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:90:13405)
	at /Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:90:11743
	at processTicksAndRejections (internal/process/task_queues.js:97:5)
	at async Promise.all (index 2)
	at async Promise.all (index 0)
/Applications/Visual Studio Code.app/Contents/Resources/app/out/bootstrap-fork.js:5
server bound
```

renanqts, 2021-02-04:
yep, package code=1.52.1-1608136922 in apt works out the box.

cpwood, 2021-02-04:
Step one - don't let it die silently from now onward should we run into this problem again:

[image]

Will move on to a root cause fix now.

flotre, 2021-02-04:
I have similar issue:

Operating system:
Ubuntu 20.04

VSCode version:
1.53.0

Pico-Go version:
1.0.4

`Error: Could not locate the bindings file. Tried: → /home/nous/.vscode/extensions/chriswood.pico-go-1.0.4/node_modules/@serialport/bindings/build/bindings.node → /home/nous/.vscode/extensions/chriswood.pico-go-1.0.4/node_modules/@serialport/bindings/build/Debug/bindings.node → /home/nous/.vscode/extensions/chriswood.pico-go-1.0.4/node_modules/@serialport/bindings/build/Release/bindings.node → /home/nous/.vscode/extensions/chriswood.pico-go-1.0.4/node_modules/@serialport/bindings/out/Debug/bindings.node → /home/nous/.vscode/extensions/chriswood.pico-go-1.0.4/node_modules/@serialport/bindings/Debug/bindings.node → /home/nous/.vscode/extensions/chriswood.pico-go-1.0.4/node_modules/@serialport/bindings/out/Release/bindings.node → /home/nous/.vscode/extensions/chriswood.pico-go-1.0.4/node_modules/@serialport/bindings/Release/bindings.node → /home/nous/.vscode/extensions/chriswood.pico-go-1.0.4/node_modules/@serialport/bindings/build/default/bindings.node → /home/nous/.vscode/extensions/chriswood.pico-go-1.0.4/node_modules/@serialport/bindings/compiled/12.18.3/linux/x64/bindings.node → /home/nous/.vscode/extensions/chriswood.pico-go-1.0.4/node_modules/@serialport/bindings/addon-build/release/install-root/bindings.node → /home/nous/.vscode/extensions/chriswood.pico-go-1.0.4/node_modules/@serialport/bindings/addon-build/debug/install-root/bindings.node → /home/nous/.vscode/extensions/chriswood.pico-go-1.0.4/node_modules/@serialport/bindings/addon-build/default/install-root/bindings.node → /home/nous/.vscode/extensions/chriswood.pico-go-1.0.4/node_modules/@serialport/bindings/lib/binding/node-v85-linux-x64/bindings.node	at bindings (/home/nous/.vscode/extensions/chriswood.pico-go-1.0.4/node_modules/bindings/bindings.js:126:9)	at Object.<anonymous> (/home/nous/.vscode/extensions/chriswood.pico-go-1.0.4/node_modules/@serialport/bindings/lib/linux.js:2:36)	at Module.u._compile (/usr/share/code/resources/app/out/vs/loader.js:3:12841)	at Object.Module._extensions..js (internal/modules/cjs/loader.js:1220:10)	at Module.load (internal/modules/cjs/loader.js:1039:32)	at Module._load (internal/modules/cjs/loader.js:932:14)	at Function.f._load (electron/js2c/asar_bundle.js:5:12738)	at Function.o._load (/usr/share/code/resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:104:27152)	at Function.p._load (/usr/share/code/resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:104:23904)	at Function.M._load (/usr/share/code/resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:94:62159)	at Module.require (internal/modules/cjs/loader.js:1079:19)	at v (/usr/share/code/resources/app/out/vs/loader.js:3:12287)	at Object.<anonymous> (/home/nous/.vscode/extensions/chriswood.pico-go-1.0.4/node_modules/@serialport/bindings/lib/index.js:14:22)	at Module.u._compile (/usr/share/code/resources/app/out/vs/loader.js:3:12841)	at Object.Module._extensions..js (internal/modules/cjs/loader.js:1220:10)	at Module.load (internal/modules/cjs/loader.js:1039:32)	at Module._load (internal/modules/cjs/loader.js:932:14)	at Function.f._load (electron/js2c/asar_bundle.js:5:12738)	at Function.o._load (/usr/share/code/resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:104:27152)	at Function.p._load (/usr/share/code/resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:104:23904)	at Function.M._load (/usr/share/code/resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:94:62159)	at Module.require (internal/modules/cjs/loader.js:1079:19)	at v (/usr/share/code/resources/app/out/vs/loader.js:3:12287)	at Object.<anonymous> (/home/nous/.vscode/extensions/chriswood.pico-go-1.0.4/node_modules/serialport/lib/index.js:2:17)	at Module.u._compile (/usr/share/code/resources/app/out/vs/loader.js:3:12841)	at Object.Module._extensions..js (internal/modules/cjs/loader.js:1220:10)	at Module.load (internal/modules/cjs/loader.js:1039:32)	at Module._load (internal/modules/cjs/loader.js:932:14)	at Function.f._load (electron/js2c/asar_bundle.js:5:12738)	at Function.o._load (/usr/share/code/resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:104:27152)	at Function.p._load (/usr/share/code/resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:104:23904)	at Function.M._load (/usr/share/code/resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:94:62159)	at Module.require (internal/modules/cjs/loader.js:1079:19)	at v (/usr/share/code/resources/app/out/vs/loader.js:3:12287)	at prepareSerialPort (/home/nous/.vscode/extensions/chriswood.pico-go-1.0.4/pymakr.js:154:9)	at activate (/home/nous/.vscode/extensions/chriswood.pico-go-1.0.4/pymakr.js:7:5)	at Function._callActivateOptional (/usr/share/code/resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:90:13728)	at Function._callActivate (/usr/share/code/resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:90:13405)	at /usr/share/code/resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:90:11743	at processTicksAndRejections (internal/process/task_queues.js:97:5)	at async Promise.all (index 2)	at async Promise.all (index 0)`

closed 2021-02-04 by cpwood

cpwood, 2021-02-04:
Looks like v1.0.5 didn't fix this after all; an issue in the build process. Addressing this under issue #11 which has been raised subsequently.

commit 601f28c1 (2021-02-04): Fix #9

Report when serialport bindings are incompatible with VS Code.
commit 967148e4 (2021-02-06): Fix #9

Pico-Go now relies on self-generated bindings.node files.