cpwood/Pico-Go #12: How to Select COM port

sakthimaan, 2021-02-05:
Not able to select the COM Port. Extension is automatically choosing the COM port. 

[image]

sakthimaan, 2021-02-05:
Note: Pico is connected to COM 4.

cpwood, 2021-02-07:
This is now next on my list, but I'll be unavailable for a few days.

snympi, 2021-02-08:
This would be useful to me as well, thanks @cpwood

cpwood, 2021-02-09:
Development is done; I just need to spend some time testing it.

Your global config (`Ctrl+Shift+G` or `Command+Shift+G`) will need to change in a way similarl to the following on macOS or Linux:

```json
	"auto_connect": false,
	"manual_com_device": "/dev/tty.usbmodem0000000000001",
```

or similar to the following on Windows:

```json
	"auto_connect": false,
	"manual_com_device": "COM3",
```

closed 2021-02-11 by cpwood

sakthimaan, 2021-02-11:
@cpwood 
Awesome.. thanks for fixing this issue. 👍

adarmaori, 2021-04-02:
When I did this and tried to connect the output was ```"Address not configured. Please go to the settings to configure a valid address or comport"```. This is the relevant part of my global settings file:
```
"auto_connect": false,
"manual_com_device": "COM5",
```
Did I do something wrong or is this a bug?

kbreezy04, 2021-08-08:
> When I did this and tried to connect the output was `"Address not configured. Please go to the settings to configure a valid address or comport"`. This is the relevant part of my global settings [...]

I also get this.  Project settings are also not respected for me.

sotnyk-lv, 2022-02-22:
Ctrl+Shift+G is opening Source Control for me.

Where or how I can find this global config file?

YayoVera26, 2022-03-13:
could you connect to your pico COM, I got the same error, address not configured.

YayoVera26, 2022-03-13:
Update, I was able to connect my pi pico to the VS, my solution:

If you have Bluetooth on your computer/laptop, try disabling it if you have it enabled this causes you to have more ports in use, by disabling it VS will recognize your pi pico as it is the only COM available.

Hope it works for you.

Anshuman-UCSB, 2022-03-21:
> 

This worked for me, seems that the settings were not respected and it kept defaulting to COM3 which was bluetooth

Elephant42, 2022-03-24:
Same here as others have reported.
Made the recommended change to global config
"auto_connect": false,
"manual_com_device": "COM7",
but keep getting the "address not configured" error.
I have several COM ports active as well as the Pico - I'm on Windows 10 Pro BTW.

commit ab783e17 (2021-02-10): Fix #12

Serial port selection