cpwood/Pico-Go #14: Visual Studio Code wont connect to Pico

deeteeakl, 2021-02-06:
(Hi! 👋  Thanks for reporting an issue! Please make sure you click the link above to view the issue guidelines, then fill out the blanks below.)

## What are the steps to reproduce this issue?
1. Connect Pico
2. Open VSCode
3. Select "Pico Console" in Terminal window

## What happens?
I get a messages below... If I click "Pico Disconnected" it tried again but again fails

Searching for boards on serial devices...
Connecting to /dev/tty.usbmodem0000000000001...
> Connection timed out. Click the "Pico Disconnected" button to try again.

## What were you expecting to happen?
Expecting to get the REPL 

## Any logs, error output, etc?
*(If it’s long, please paste to https://gist.github.com and insert the link here)*
Searching for boards on serial devices...
Connecting to /dev/tty.usbmodem0000000000001...
> Connection timed out. Click the "Pico Disconnected" button to try again.

## Any other comments?
I also have Thonny and tested with that and it works without issue

## Which versions of software are you using?

**Operating system:**
Mac OS 11.2 Big Sur

**VSCode version:**
1.53

**Pico-Go version:**
v1.0.7

## Which hardware are you using?
Pi Pico

E.g. Raspberry Pi Pico or the name of another RP2040 board.

cpwood, 2021-02-06:
Hi @deeteeakl - I've just tested this on two Macs (an Intel MacBook Pro and also an M1 Mac Mini) and I'm not able to reproduce this. A few other Mac users have reported 1.0.7 working fine as part of #11 too.

**That said,** this is *exactly* the issue I was running into as part of resolving #11. I found the issue was down to the pre-built binaries in the serialport npm package and ended up building my own and including them in 1.0.7 to move forward.

So I'm a little confused why you're still running into it...!

To rule out the basics... any chance something else is connected to the Pico, e.g. Thonny, therefore Pico-Go can't connect? Any better after a reboot (sorry)?

deeteeakl, 2021-02-06:
Hi,

Yes I was wondering that also.  I have rebooted a couple of times but still not dice.  I have rebooted tried with VSCode, wont connect, then open Thonny and try there and it works.

Are there any diagnostics I can get?

cpwood, 2021-02-06:
Will have a ponder on what we can try in that regard. Still getting used to the code myself after inheriting it from Pymakr at this point. 

I do get the sense that serialport has been a constant issue for the folks over at PyCom too, unfortunately.

cpwood, 2021-02-06:
One thought.. does the (ultimately unreleased) 1.0.6 work on your machine? You'll find a link to it in #11. The irony would be that it works for you but didn't for everyone else!

fredbouchard, 2021-02-06:
Hi @cpwood 

First, thank you for this VSCode plugin! 🙌

I installed the version `1.0.6` like you suggest and same as @deeteeakl it won't connect.  

I can connect my device with `Thonny`: ✅

[image]

If you want my help to try things or help you with the debug process let me know.

## More details about my setup
```bash
VSCode: v1.53.0
OS: Linux x64 5.8.0-7642-generic (Pop!_OS 20.10 Nvidia)
PICO Firmware: rp2-pico-20210202-v1.14.uf2 (stable)
Node: v14.15.4 (lts/fermium)
Python: v3.8.6
Pip: v21.0.1
```

deeteeakl, 2021-02-06:
Hi @cpwood 

I just opened my laptop to start work on my project and found that the pico-go extension was saying vs code needed a restart.  I restarted and hey presto!  All working perfectly now.

Thanks so much for the extension, your web site etc.  Really looking forward to getting stuck in.

fredbouchard, 2021-02-06:
I'm back on `1.0.7` and did not have the same luck as @deeteeakl. Actually, on my end it keep trying to connect and never return a timeout.

deeteeakl, 2021-02-06:
Looks like I spoke too soon.  I had the connection working and then started working through setting up the environment by selecting the stub and linter.  When I re-started VSCode i get repeated errors displaying the following:

The terminal process "node" /path to extension/chriswood.pico-go-1.0.7/TerminalExec.js" '1829' terminated with exit code 1

the message repeats with different line numbers (1557, 1504, 2156, 2261, etc)

fredbouchard, 2021-02-06:
Well... I'm getting crazy. 

1. I un-installed the plugin
2. Re-installed it
3. Restart VSCode
4. Still the issue
5. Start Thonny on top of that (both VSCode and Thonny errored-out obviously)
6. Close Thonny
7. Restart VSCode
8. Pico Console: `>>>` Connection success
9. `Run` Flash.py script
10. Success blinking light 🎉

[image]

cpwood, 2021-02-07:
@fredbouchard @deeteeakl can you let me know if you feel further work is required, or is all well now?

Seems it's been a little unpredictable for you at the very least, so if you have any insights on potential root causes or ways to reproduce the problems consistently, I'm all ears.

deeteeakl, 2021-02-07:
Unfortunately its not working for me at all and I am back to using Thonny again.  My update above provides some details.  I tried deleting the extension and adding again but no joy.  I played around a bit with turning on logging to try and trace the issue but did not get far.

cpwood, 2021-02-09:
Not sure if this will help, however I've put together a [document](https://github.com/cpwood/Pico-Go/blob/develop/connection-issues.md) on steps we can work through while diagnosing connection issues.

cpwood, 2021-02-11:
Going to close off this issue at this point as it's not something I'm able to reproduce.

closed 2021-02-11 by cpwood

MariaRigaki, 2021-04-14:
I know this is a closed issue but I'm not sure how else to help @fredbouchard 
Since you are running Linux, chances are that you don't have permission to connect to `/dev/ttyACM0`.

MariaRigaki, 2021-04-14:
I know this is a closed issue but I'm not sure how else to help @fredbouchard 
Since you are running Linux, chances are that you don't have permission to connect to `/dev/ttyACM0`. 

The solution that worked for me was to add my user to the `dialout` group that owns the `/dev/ttyXXX` and then restart.

fredbouchard, 2021-04-14:
@MariaRigaki Thank you! I'll keep that in mind if I experienced this issue again.

djhunter67, 2021-06-08:
SOLVED!!!!!  Running VSCode as sudo solved this issue. I do not know how to run any script that I make but the REPL in VSCode works and the Flash function is running as expected.