cpwood/Pico-Go #15: Support Linux arm64

cpwood, 2021-02-07:
Important for supporting Raspberry Pi 4. It's `serialport`'s bindings that need addressing.

cpwood, 2021-02-07:
Linux arm64 bindings for `serialport` have now been built successfully but need testing.

cpwood, 2021-02-07:
Folks! If anybody has an arm64 Raspberry Pi and would like to try a specially-compiled version of 1.0.7, you can grab it [here](https://cwood.blob.core.windows.net/temp/pico-go-1.0.7.vsix).

> **NOTE:** I've **not** tested this myself as I don't currently have an arm64 Raspberry Pi with an arm64 OS available. So I'm releasing this "blind" and need user feedback / help on problems.

I'm not planning on a 32-bit ARM version at the moment since `serialport` is 64-bit only.

Apologies if I'm slow to respond in the coming days; real life is dominating proceedings!

cc: @andypiper

thijsputman, 2021-02-08:
Just tried the specially-compiled 1.0.7 version noted above. I'm afraid I'm seeing the exact same behaviour as described in https://github.com/cpwood/Pico-Go/issues/17...

Can you perhaps tag it with a different version number to differentiate it from the Marketplace one? Although I did fully remove the Marketplace variant and restart VS Code a couple times, I'm unable to 100% validate VS Code is now using the specially-compiled version.

Running Ubuntu 20.10 ARM64 on a Raspberry Pi 4 (via "Remote - SSH", but shouldn't an issue).

cpwood, 2021-02-08:
Thanks @thijsputman! All good recommendations.

You can check you're definitely running the ARM64 version by looking at the content of the extension's folder on your hard disk. You should see folders ending with `arm64` in the `node_modules/@serialport/bindings/lib/binding` folder.

I aim to get a Raspberry Pi 4 to test this on myself in a short while.

thijsputman, 2021-02-09:
Just checked and it indeed has ARM64 bindings:

1. `node-v80-linux-arm64`
2. `node-v85-linux-arm64`

So I guess the error I'm getting is genuine... 🙂

Let me know if there's anything else I can do to help!

cpwood, 2021-02-09:
OK, thanks for the feedback. FWIW, I've just uploaded [1.0.8-arm64](https://cwood.blob.core.windows.net/temp/pico-go-1.0.8-arm64.vsix) (which is a pre-release) but there's no actual change as far as Linux ARM64 is concerned (the change is macOS ARM64 which I _can_ test and appears to be OK at first glance).

I best order my RP4 for testing! 😉

cpwood, 2021-02-09:
OK, RP4 ordered. Will get to this at some point in the coming week when I fancy a distraction.

cpwood, 2021-02-10:
RP4 received! I managed to get this working:

[image]

The board wasn't plugged in initially, hence the `No boards found on USB` message.

Here's a [download link](https://cwood.blob.core.windows.net/temp/pico-go-1.0.8-beta1.vsix) for another beta - does this work any better for you? If so, it seems cross-compiling for ARM64 on an x64 machine is a bit dodgy (which is what I was attempting previously).

cpwood, 2021-02-11:
Will put this in the 1.0.8 release. Closing this issue.

closed 2021-02-11 by cpwood

thijsputman, 2021-02-11:
@cpwood, many thanks for the super quick turnaround!

I'm afraid it still isn't working for me though...

Fished out the pertinent error message (from my current installation; running `1.0.8` from the Marketplace):

```
Error: Could not locate the bindings file. Tried:
 → /home/thijs/.vscode-server/extensions/chriswood.pico-go-1.0.8/node_modules/@serialport/bindings/build/bindings.node
 → /home/thijs/.vscode-server/extensions/chriswood.pico-go-1.0.8/node_modules/@serialport/bindings/build/Debug/bindings.node
 → /home/thijs/.vscode-server/extensions/chriswood.pico-go-1.0.8/node_modules/@serialport/bindings/build/Release/bindings.node
 → /home/thijs/.vscode-server/extensions/chriswood.pico-go-1.0.8/node_modules/@serialport/bindings/out/Debug/bindings.node
 → /home/thijs/.vscode-server/extensions/chriswood.pico-go-1.0.8/node_modules/@serialport/bindings/Debug/bindings.node
 → /home/thijs/.vscode-server/extensions/chriswood.pico-go-1.0.8/node_modules/@serialport/bindings/out/Release/bindings.node
 → /home/thijs/.vscode-server/extensions/chriswood.pico-go-1.0.8/node_modules/@serialport/bindings/Release/bindings.node
 → /home/thijs/.vscode-server/extensions/chriswood.pico-go-1.0.8/node_modules/@serialport/bindings/build/default/bindings.node
 → /home/thijs/.vscode-server/extensions/chriswood.pico-go-1.0.8/node_modules/@serialport/bindings/compiled/12.18.3/linux/arm64/bindings.node
 → /home/thijs/.vscode-server/extensions/chriswood.pico-go-1.0.8/node_modules/@serialport/bindings/addon-build/release/install-root/bindings.node
 → /home/thijs/.vscode-server/extensions/chriswood.pico-go-1.0.8/node_modules/@serialport/bindings/addon-build/debug/install-root/bindings.node
 → /home/thijs/.vscode-server/extensions/chriswood.pico-go-1.0.8/node_modules/@serialport/bindings/addon-build/default/install-root/bindings.node
 → /home/thijs/.vscode-server/extensions/chriswood.pico-go-1.0.8/node_modules/@serialport/bindings/lib/binding/node-v72-linux-arm64/bindings.node
```

Running `node -v` on the Raspberry Pi produces `v14.15.5` (Node is installed via `snapd`; perhaps that's part of the issue?)

VS Code version information:
```
Version: 1.53.1 (user setup)
Commit: 5d424b828ada08e1eb9f95d6cb41120234ef57c7
Date: 2021-02-08T23:29:42.785Z
Electron: 11.2.1
Chrome: 87.0.4280.141
Node.js: 12.18.3
V8: 8.7.220.31-electron.0
OS: Windows_NT x64 10.0.19042
```

Running via the "Remote - SSH" extension, so disregard the Windows part 🙂

It appears somehow "version confused" between the Node provided by VS Code and the one installed on the machine itself?

cpwood, 2021-02-11:
I can see this in the logs:

`node-v72-linux-arm64`

which is completely the wrong version for Electron 11.2.1. It should be looking for v85.

V72 is something like Electron 4 or 5!

Unfortunately, this isn't something Pico-Go does; it's within serialport. 

I'll see if I can figure out why it's getting so confused.

cpwood, 2021-02-11:
I'm not familiar with the remote ssh VS Code stuff. Am I right in saying that the UI's on Windows on your case, and the host is the RP?

Are you able to run Code "normally" on the RP via VNC or directly? Does that work OK?

Just trying to figure out whether it's something in and around the remote approach. Maybe the host bit uses a really old version of Electron?

cpwood, 2021-02-11:
This probably has something to do with it..

https://code.visualstudio.com/docs/remote/troubleshooting#_extensions-that-ship-or-acquire-prebuilt-native-modules-fail

cpwood, 2021-02-11:
For now, I'll make a change to alert users that the Remote SSH extension isn't supported, so it doesn't appear to just do nothing. Unfortunately, making it work with the extension will probably require shipping both Node and Electron bindings and then moving files about depending upon the environment context.

thijsputman, 2021-02-12:
> This probably has something to do with it..
> 
> https://code.visualstudio.com/docs/remote/troubleshooting#_extensions-that-ship-or-acquire-prebuilt-native-modules-fail

That looks like it yes...

I'm not super familiar with VS Code extension development, but wouldn't mind taking a stab at this issue in the next couple of weeks.

The document mentions "dynamically acquire" the required binaries. Do you reckon it would be possible to compile them on demand on the specific device/architecture (basically running `node-gyp` I guess?), or is that a can of worms I shouldn't open?

cpwood, 2021-02-12:
For now, can you try an `npm rebuild serialport` from within the extension's folder on the RP? That should rebuild it in for plain old Node for arm64 and in theory move things along past this particular issue. 

I'm curious whether the nature of the extension (part UI, part back-end) means that we'd then run into more insurmountable problems after this one. So we should probably try and hack it into working shape before finessing anything more permanent.  🙂

andypiper, 2021-02-12:
Is this supposed to work on Raspi 400 - I am using the default arm32 build they ship as Raspberry Pi OS.

```
$ uname -a
Linux pi400 5.10.11-v7l+ #1399 SMP Thu Jan 28 12:09:48 GMT 2021 armv7l GNU/Linux
```

I see an error about not being compatible with Linux (arm), rather than (arm64)

cpwood, 2021-02-12:
@andypiper I'll try to see if there's any mileage in a 32-bit version but it's all down to the `serialport` dependency and its reliance on a 64-bit version of Node. It's not supported on Windows or Mac 32-bit platforms for the same reason.

I do wonder what the story is with Raspberry Pi OS x64. I'm running it on my RP4 and works great, but it feels like it's "drifting" as a project, somehow. As you say, it's not one of the default installation options despite it being an ARM64 processor!

thijsputman, 2021-02-12:
> For now, can you try an `npm rebuild serialport` from within the extension's folder on the RP? That should rebuild it in for plain old Node for arm64 and in theory move things along past this particular issue.

@cpwood, that worked!

Running `npm rebuild serialport` didn't do anything. I had to run a full `npm rebuild` to get `serialport` to recompile. Additionally, the Node version installed on the Raspberry Pi should – of course – match the one bundled with VS Code for this to work.

The extension appears to work just fine now: I can use REPL and pressing "▶&#xfe0e; Run" successfully runs the currently open Python-file on the microcontroller.

cpwood, 2021-02-12:
@thijsputman that's great news! I'll create a new issue do do a proper change to address this requirement. Thanks for trying that out!