cpwood/Pico-Go #16: Support Apple M1 (ARM64)

cpwood, 2021-02-07:
`serialport`'s bindings need addressing and also need to be able to use `electron-rebuild` for M1/ARM64 on Darwin.

cpwood, 2021-02-07:
Just attempted to do a build for Darwin arm64 with my Git Action and got:

```
/Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk/usr/include/sys/cdefs.h:807:2: error: Unsupported architecture
#error Unsupported architecture
```

so GitHub isn't ready for this yet. It [does seem `electron-rebuild` *is* ready](https://www.electronjs.org/blog/apple-silicon) though.

studer, 2021-02-08:
> Just attempted to do a build for Darwin arm64 with my Git Action and got:
>
> ```
> /Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk/usr/include/sys/cdefs.h:807:2: error: Unsupported architecture [...]

Are you using the right SDK for arm64 ? Mine's at `/Library/Developer/CommandLineTools/SDKs/MacOSX11.1.sdk`

cpwood, 2021-02-08:
It's using one of GitHub's hosted build environments and I'm not important or special enough to have access to the `macos-11.0` machines 😉 Only have `mac-latest`. 

Will consider managing M1 Macs manually instead of through build automaton in the interim.

cpwood, 2021-02-09:
Will also investigate this..

https://www.jessesquires.com/blog/2020/01/06/selecting-an-xcode-version-on-github-ci/

cpwood, 2021-02-09:
We have lift-off! 🚀 I've managed to generate some Darwin arm64 binaries for `serialport`. The secret sauce (over in [Bindings-Builder's build action](https://github.com/cpwood/Bindings-Builder/blob/main/.github/workflows/build.yml)) was this:

```
    - name: Ensure latest macOS SDK is used
      if: matrix.os == 'macos-latest'
      shell: zsh {0}
      run: |
        sudo xcode-select --switch "$(find /Applications -mindepth 1 -maxdepth 1 ! -type l | grep "Xcode_[\.0-9]*app" | sort -V | tail -1)/Contents/Developer"
        sudo rm -Rf /Library/Developer/CommandLineTools/SDKs/*
```

Will get this into a build of Pico-Go soon.

cpwood, 2021-02-09:
Have to say, it's looking good on Silicon!

[image]

Here's [1.0.8-arm64](https://cwood.blob.core.windows.net/temp/pico-go-1.0.8-arm64.vsix) (which is basically a pre-release) for anybody to try.

Look forward to any feedback!

studer, 2021-02-09:
Nice work, thank you ! 👍 😎

I was able to install it, but I still cannot run any commands.
I always receive the following error :

> Command 'Pico-Go > Global settings' resulted in an error (command 'pymakr.globalSettings' not found)

In the Debug Console I can see the following error message :

> Error: Cannot find module 'serialport'

cpwood, 2021-02-10:
@studer can you check whether this folder exists within extension's folder in `~home/.vscode/extensions`?

`node_modules/@serialport/bindings/lib/binding/node-v85-darwin-arm64`

Or even whether the `@serialport` folder exists!

It's definitely in the visx file (you can rename it to .zip and then look inside).

FWIW, all I did to get the screenshot above was uninstall any existing Pico-Go extension, install the new one from the 1.0.8 visx and then restart VS Code.

Have you done anything *beyond* that? Just wondering if there's a vulnerability along the lines of "if I then do x, it all breaks".

studer, 2021-02-10:
Yep, the folder is there and contains two files : 

 - bindings.node: `Mach-O 64-bit bundle arm64`
 - info.json:     `JSON data`

But I've managed to make it work !

I've uninstalled any previous version of pico-go, restarted VS Code and reinstalled the vsix.
Then, after checking the debug console again, I saw the following error message (and no mention of `Cannot find module 'serialport'` anymore) : 

> Error: ENOENT: no such file or directory, open '/Users/studer/Library/Application Support/Code/User/pico-go.json'

The directories `Code/User/` were not existant on my machine. So as soon as I created them and restarted VS Code, pico-go started working 🚀 !

cpwood, 2021-02-11:
Interesting, I'll see if I can ensure that folder is created by the extension if it doesn't already exist.  Thanks!

cpwood, 2021-02-11:
OK, that folder will now get created if it doesn't already exist. I'll close off this issue at this point.

closed 2021-02-11 by cpwood

hrueger, 2021-11-14:
> The secret sauce (over in Bindings-Builder's build action) was this:

Thanks so much @cpwood!
This also helped me with my `Electron` project which would not build for arm64.

commit 3af7b769 (2021-02-11): Fix #16

Ensure settings folder exists. Main body of work done by Binding-Builder commits.