cpwood/Pico-Go #19: developing for VS Code 1.53

Naohiro2g, 2021-02-08:
## What are the steps to reproduce this issue?
1. Follow the instructions at Developing in README.md.
2. Run 'Launch Extension' to open new window for extension testing.

## What happens?
"This version of Pico-Go is incompatible with VSCode 1.53.0."

Error: The module '......./Pico-Go/node_modules/@serialport/bindings/build/Release/bindings.node'
was compiled against a different Node.js version using NODE_MODULE_VERSION 72.
This version of Node.js requires NODE_MODULE_VERSION 85. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).

It seems still using electron 6.1.2

## What were you expecting to happen?
npm rebuild to make correct extension.

I was also testing Bindings-Builder to clone native_modules into Pico-Go directories with GitHub actions and once I got the extension working properly but I cannot reproduce the process. 

## Which versions of software are you using?
**Operating system:**
Darwin x64 18.7.0 or macOS mojave
**VSCode version:**
1.53.0
**Pico-Go version:**
1.0.7

## Which hardware are you using?
MacBook Air Core i7

cpwood, 2021-02-08:
In terms of building `serialport` for _Electron_ the steps are slightly different to what you're doing with `npm rebuild`.

You're playing with stuff I _literally_ wrote over the weekend, isn't fully-documented yet, and some of which only exists on Pico-Go's `develop` branch at the moment! 😄 

**Bindings-Builder:**

1. use `electron-rebuild` to rebuild the component for the Electron versions of your choice as defined within `package.json`:

```json
  "codeVersions": [
    {
      "code": "1.54.0",
      "electron": "11.2.2",
      "modules": 85
    },
    {
      "code": "1.53.0",
      "electron": "11.2.1",
      "modules": 85
    },
    {
      "code": "1.52.0",
      "electron": "9.3.5",
      "modules": 80
    },
    {
      "code": "1.51.0",
      "electron": "9.3.3",
      "modules": 80
    }
  ]
```

2. add the recompiled bindings.node file to the `native_modules/node-v{XX}-{platform}-{arch}` folder at the root of the Pico-Go repo;
3. commit the code to the `develop` branch of Pico-Go and push.

The whole series of steps is designed to be run on a GitHub-hosted build servers through the GitHub Action. The action will do a matrix build on Win32, Linux and Darwin hosted machines. It'll do x64 for each of those platforms and also amd64 on Linux.

**Pico-Go**

4. pull the commits with the new `bindings.node` files from `develop`;
5. copy those files from `native_modules` into `node_modules/@serialport/bindings/bin/binding` folder after every `npm install` or when running the `copy-native` NPM script..

closed 2021-02-08 by cpwood

Naohiro2g, 2021-02-09:
Thanks for the explanation.  (GitHub action at Pico-Go is not complete, right? I couldn't run it.)

I had not beeing aware the GitHub action at Pico-Go, so did it with the action at Bindings-Builder again.

I deleted the  v85-darwin-x64 directory at Pico-Go/develop and kicked the action at Bindings-Builder.
The new native was created successfully and I pulled it.
Then, I've lost at the last section.

Do I have to mkdir ```node_modules/@serialport/bindings/bin/binding``` ?
Anyway, it was good for me to learn about ABI, node, npm, GitHub actions, node-serialports, and many more.

My question is why native node-module-v85 is not working with VS Code 1.53. I was thinking v85 built with Electron 11.2.1 can make it.  I have to learn to know around node build/install system much more.

cpwood, 2021-02-09:
`node_modules/@serialport/bindings/bin/binding` *should* already exist since `serialport` ships with bindings for Windows, Mac and Linux (albeit ones for Node as opposed to Electron). They're located in the `binding` folder.

Using `electron-rebuild` with an Electron version of `11.2.1` *should* result in a valid v85 binding being created.

cpwood, 2021-02-09:
FYI - I've now finished my work on Bindings-Builder and have done some [initial documentation](https://github.com/cpwood/Bindings-Builder/blob/main/README.md).

Naohiro2g, 2021-02-10:
```npm install``` creates ```node_modules``` using ```package.json```, right? Then ```@serialport/bindings/bin/binding``` should be there.

Is this the right location for the new file?  Now, I don't have it, something went wrong with npm install.
```Pico-Go/node_modules/@serialport/bindings/bin/binding/node-v85-darwin-x64/bindings.node```

I will try it again from now.

BTW, pyMakr released the update but it still didn't work.