cpwood/Pico-Go #28: Terminal exits repeatedly with exit code: 1

greysou1, 2021-02-13:
https://user-images.githubusercontent.com/29304096/107855749-25751f00-6e4a-11eb-99f3-beec93aaeac3.mov

(Hi! 👋  Thanks for reporting an issue! Please make sure you click the link above to view the issue guidelines, then fill out the blanks below.)

## What are the steps to reproduce this issue?
1. I just installed this extension and the terminal goes in a loop of exiting and starting
2.
3.

## What happens?
It keeps saying "execvp(3) failed.: No such file or directory" 

## What were you expecting to happen?
able to use the terminal to communicate with the PICO and get working with it

## Any logs, error output, etc?
*(If it’s long, please paste to https://gist.github.com and insert the link here)*
"execvp(3) failed.: No such file or directory" 
"The terminal process "node '/Users/prudvikamtam/.vscode/extensions/chriswood.pico-go-1.0.8/terminalExec.js' '1987' terminated with exit code: 1.

## Any other comments?
nope

## Which versions of software are you using?

**Operating system:**
macOS Big Sur Version 11.1
**VSCode version:**
Version: 1.53.2
**Pico-Go version:**
Version: 1.53.2

## Which hardware are you using?
Raspberry Pi Pico

cpwood, 2021-02-17:
Thanks for the detailed information!

I'm not familiar with the exception in detail, however other incidents relating to that error message seem to allude to the default terminal or Node being misconfigured in VS Code, somehow.

Pico-Go is based upon Pymakr:

https://forum.pycom.io/topic/6015/mac-vs-code-pymakr

And these are other terminal-related issues (albeit not with Pico-Go, I acknowledge):

https://github.com/vercel/hyper/issues/3703#issuecomment-500095273
https://github.com/platformio/platformio-atom-ide-terminal/issues/5#issuecomment-222671359

Is there any chance you've seen other Node or Terminal-related issues on your machine?

And to ask the standard tech support question: don't suppose a reboot fixes it?

cpwood, 2021-02-17:
Looking into this some more, Pico-Go is trying to create the Pico Console by calling `node terminalExec.js`. If that fails, it will repeatedly retry (which I've just reproduced by making it fail on purpose).

Questions..

1) If you type `node --version` at your usual terminal window, does it work?
2) Do you have anything that might be stopping Pico-Go from opening up a TCP port on 127.0.0.1? The port is randomised.

greysou1, 2021-02-18:
Whew took me half a day but finally figured it out!

First I checked that my node version was 10.x.x so I updated in hopes it'd work but nope no luck but after going through the links you've provided I've understood that VSCode wasn't able to find the path of node for some reason.
This was because while it was able to reference to the node path in default env when your code was opening a remote env it couldn't find the node path. 

So a **simple fix** I found in this [link](https://github.com/microsoft/vscode/issues/70248#issuecomment-508740492) this was to go settings and search for "terminal.integrated.inheritEnv" and turn it on. That's it!! 
[image]

closed 2021-02-18 by greysou1