cpwood/Pico-Go #32: Not resetting code in Pico?

m0nux, 2021-02-17:
## What are the steps to reproduce this issue?
1.Load the Blink python example code.
2. Press Run
3. Press Run again multiple times

## What happens?
LED seems to get multiple Timers trying to toggle it resulting in an odd LED flashing pattern. 

## What were you expecting to happen?
LED to flash at the frequency specified. 

## Any logs, error output, etc?
No errors shown
*(If it’s long, please paste to https://gist.github.com and insert the link here)*

## Any other comments?
If for example, change the frequency to 10, in order to get correct flashing you have to unplug USB and plug back in the press Run again.
This does not happen in Thonny

## Support info

Copy this from the `Pico-Go > Extra > Get support info` option in the command palette:

```
Pico-Go:      1.1.0
VS Code:      1.53.2
Electron:     11.2.1
Modules:      85
Node:         12.18.3
Platform:     win32
Architecture: x64
Board:        Raspberry Pi Pico with RP2040
Firmware:     v1.14 on 2021-02-02 (GNU 9.3.0 MinSizeRel)
```

cpwood, 2021-02-17:
Thanks for reporting this! Does this help at all?

http://pico-go.net/docs/help/settings/#safe_boot_on_upload

Try setting it to true?

—
auth/ADKRWGB2ZF5OGEZH3HFU453S7QD65ANCNFSM4XY3VHZA>.

cpwood, 2021-02-17:
Sorry, back at my machine now and can reproduce the problem. Can confirm `safe_boot_on_upload` doesn't work too.

cpwood, 2021-02-17:
FWIW, `Ctrl+D` will reset everything without unplugging the USB cable.

Current gut feeling is to do two things:

1. New configuration option to do a soft reset before running code (configured "off" by default);
2. Menu option to do resets (hard / soft) in the Pico-Go command palette.

The reason for _not_ having this "on" by default is that you should still be able to incrementally debug without losing state in between highlighted lines of code.

Code to initiate a soft reset is:

```python
import machine
machine.soft_reset()
```

Full reset:

```python
import machine
machine.reset()
```

m0nux, 2021-02-17:
Thank you for checking. I'll try the Ctrl D option.

cpwood, 2021-02-17:
Just for my own ease of lookup later on or for anybody else that's curious, the `blink.py` code referred to is this:

```python
from machine import Pin, Timer

led = Pin(25, Pin.OUT)
tim = Timer()
def tick(timer):
    global led
    led.toggle()

tim.init(freq=2.5, mode=Timer.PERIODIC, callback=tick)
```

https://github.com/raspberrypi/pico-micropython-examples/blob/master/blink/blink.py#L1-L9

cpwood, 2021-02-17:
This fix has now been coded and works fine, though I did have a change of heart on approach. This will be included in `v1.1.1`.

## Reset Status Bar Button

This performs a soft reset.

[image]

## Keyboard shortcut

When focused on the REPL window, `Ctrl+D` or `Command+D` has always performed a soft reset. However, you can now use `Ctrl+Shift+D` or `Command+Shift+D` to perform a soft reset from _any_ part of VS Code (e.g. when focused in the text editor window).

## Command Palette

Now has soft and hard reset options.

[image]

## Soft Reset on Run

* If a _file_ is being run, a soft reset _will_ be performed in advance of the code being executed.
* If a _selection of code_ is being run, a soft reset _won't_ be performed.

m0nux, 2021-02-18:
That all makes a great deal of sense Chris - a good solution.
Thanks for your efforts with this extension, it made my life simpler!
All the best
Julian

cpwood, 2021-02-20:
Re-opening until the change is released for download.

cpwood, 2021-02-24:
Thanks for your patience. This is still coming, but I really want to get #35 finished before it's released.

m0nux, 2021-02-25:
No pressure from me Chris - nothing worth having is ever easy or quick :)

commit 4faa603d (2021-02-17): Fix #32

Soft and hard resets via UI; soft reset in advance of Run.
closed 2021-03-04 by cpwood