cpwood/Pico-Go #34: REPL while running

sstaub, 2021-02-19:
# Feature Request 🚀

## Is your feature request related to a problem? Please describe.
If a program is running you have to stop it before you can use the REPL, this can be done with pressing RUN again or CTRL+C

## Describe the solution you'd like
A dedicated button naming STOP or REPL would be more intuitive, maybe also a symbol (stop) when execution is stopped.

cpwood, 2021-02-19:
Thanks, @sstaub! This is actually a bug as it _should_ already be doing this! Looks like a bug inherited from Pymakr.

The good news is that the code is now complete and it will be included in version 1.1.1.

cpwood, 2021-02-24:
Thanks for your patience. This is still coming, but I really want to get #35 finished before it's released.

sstaub, 2021-02-24:
No problem, I will start a pico project in a couple of weeks because I have to finish another project first.

closed 2021-03-04 by cpwood