cpwood/Pico-Go #36: Different behaviors between Run and Softreset(CTRL-D)

sstaub, 2021-02-20:
## What are the steps to reproduce this issue?
1. Example simple blinky with sleep
```
from machine import Pin
import utime

led = Pin(25, Pin.OUT)

while (1):
	led.toggle()
	utime.sleep(1)
```
- RUN works, you have no access to REPL
- while RUN you can't make a soft reboot 
- another RUN stops, you have access to REPL
- the next RUN restarts
this behavior is expected (see also #33) 
2. Example simply blink with timer
```
from machine import Pin, Timer
import utime

led = Pin(25, Pin.OUT)
tim = Timer()

def tick(timer):
	global led
	led.toggle()

tim.init(freq=2, mode=Timer.PERIODIC, callback=tick)
```
- first RUN starts and you have access to REPL, which is different to the the first example because there is no loop and only the Timer starts
- next RUN starts the Timer again but it seems there is starting a new instance of Timer because with every RUN the led blinks more chaotic. There is a soft reboot(CTRL-D)  necessary to RUN it correctly

## What were you expecting to happen?
The behavior of RUN, when running a file (not a line!) should include a soft reboot to avoid running instances in the background.

## Support info

Copy this from the `Pico-Go > Extra > Get support info` option in the command palette:
This does not work, I get no info
version is 1.1.0
vscode-insider 1.54.0 insider

cpwood, 2021-02-20:
Really appreciate you taking the time to provide all that detail.

The good news is that this feels like a hybrid of #32 and #34, both of which are already fixed and are awaiting release. 

The original user closed #32, it seems, which is probably why you've not noticed it. I'll re-open it for visibility until it's released. 

Just finishing up some unrelated code refactoring and then it'll be available for download via the marketplace.

closed 2021-02-20 by cpwood

sstaub, 2021-02-20:
This behavior is very close to #32, sorry I didn't read that issue. Thank you for this nice extension, I can now delete the Thonny IDE because I'm familiar with VSC using PlatformIO.

sstaub, 2021-03-05:
In 1.2 it is much better because restart button makes easy to avoid the timer problem. Maybe restart should done automatically before run.

cpwood, 2021-03-06:
@sstaub that should've been included in 1.2 but I've just tested it and it didn't work 😞 Looks like the change got lost in the big refactor of code I did.

I've just reimplemented it; will be in the next version.