cpwood/Pico-Go #38: Can't configure new project: "pymakr.initialise failed"

joelburton, 2021-02-23:
## What are the steps to reproduce this issue?
1. Have MacOS w/Python, Node, VSCode already installed and known-working for other things
2. Install Pico-Go 1.1.0
3. Make empty folder; configure project

## What happens?
"Command 'Pico-Go > Configure project' resulted in an error (Running the contributed command: 'pymakr.initialise' failed.)"

## What were you expecting to happen?
Configuration to work

## Any logs, error output, etc?
*(If it’s long, please paste to https://gist.github.com and insert the link here)*

## Any other comments?

```
Pico-Go:      1.1.0
VS Code:      1.53.2
Electron:     11.2.1
Modules:      85
Node:         12.18.3
Platform:     darwin
Architecture: x64
Board:        Raspberry Pi Pico with RP2040
Firmware:     v1.14 on 2021-02-02 (GNU 9.3.0 MinSizeRel)
```

I'll note that I have Node v14.15.5 installed; the 12.18.3 here might be VSCode's internal Node, perhaps?

```
~ % node
Welcome to Node.js v14.15.5.
Type ".help" for more information.
```

joelburton, 2021-02-23:
I'm very excited to use your library, and am delighted with your documentation. Any help anyone can offer here is much appreciated.

cpwood, 2021-02-24:
Thanks - I think this is probably related to #40, so will be nice and easy to resolve, if so. (Famous last words!)

Will get to it very soon. Focusing on #35 at the moment which is taking a bit of time.

joelburton, 2021-02-24:
It doesn't look like this is related to a keyboard shortcut -- I'm trying to invoke configure using the command palette (cmd-shift-p on a Mac). I changed my pico-configure to a different shortcut and no error.

I did find a traceback in the "Log (Extension Host)":

```
[2021-02-24 12:56:01.958] [exthost] [error] TypeError [ERR_INVALID_ARG_TYPE]: The "path" argument must be of type string. Received null
	at validateString (internal/validators.js:120:11)
	at Object.join (path.js:1039:7)
	at StubsManager.addToWorkspace (/Users/joel/.vscode/extensions/chriswood.pico-go-1.1.0/lib/stubs/stubs-manager.js:60:28)
	at /Users/joel/.vscode/extensions/chriswood.pico-go-1.1.0/pymakr.js:53:28
	at b._executeContributedCommand (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:86:88517)
	at b.$executeContributedCommand (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:86:88849)
	at p._doInvokeHandler (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:89:10276)
	at p._invokeHandler (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:89:9968)
	at p._receiveRequest (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:89:8638)
	at p._receiveOneMessage (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:89:7440)
	at /Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:89:5568
	at v.fire (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:55:1836)
	at d.fire (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:63:15515)
	at /Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:104:29104
	at v.fire (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:55:1836)
	at d.fire (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:63:15515)
	at t._receiveMessage (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:63:20765)
	at /Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:63:17659
	at v.fire (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:55:1836)
	at u.acceptChunk (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:63:12880)
	at /Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:63:12228
	at Socket.m (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:104:12375)
	at Socket.emit (events.js:315:20)
	at addChunk (_stream_readable.js:295:12)
	at readableAddChunk (_stream_readable.js:271:9)
	at Socket.Readable.push (_stream_readable.js:212:10)
	at Pipe.onStreamRead (internal/stream_base_commons.js:186:23) pymakr.initialise
	```

lewixlabs, 2021-03-02:
@cpwood great extension! 👍
I follow this issue, I hope this issue will be solved to develop with pico & vscode 👌
Thanks!

cpwood, 2021-03-03:
If you're experiencing this problem, can you confirm you've created an empty folder **and have opened it in VS Code** before running the command?

The stack trace above (thanks @joelburton!) seems to indicate that the extension can't get the path to the current project (i.e. the folder isn't open in VS Code).

joelburton, 2021-03-03:
I'm not 100% sure if that was my problem (I haven't kept my editor open since I hit that bug), but starting from scratch with a new, blank folder opened up, I am able to configure it. 

Thanks, @cpwood! I admire your dedication to this codebase and appreciate your help.

cpwood, 2021-03-03:
No worries, @joelburton - glad it's working! 👍 

I've added a small change so that Pico-Go will give some guidance if you try and run that command but don't have a project open.

lewixlabs, 2021-03-04:
> If you're experiencing this problem, can you confirm you've created an empty folder **and have opened it in VS Code** before running the command?
>
> The stack trace above (thanks @joelburton!) seems [...]

Yes, I confirm, following those steps extensions works! 👍
For info: I'm using VsCode v1.53.2 from macOS Catalina 10.15.7.

Thanks @cpwood

closed 2021-03-04 by cpwood

Kingjulien55, 2022-06-10:
> ## What are the steps to reproduce this issue?
> 1. Have MacOS w/Python, Node, VSCode already installed and known-working for other things
> 2. Install Pico-Go 1.1.0
> 3. Make empty folder; configure [...]

> It doesn't look like this is related to a keyboard shortcut -- I'm trying to invoke configure using the command palette (cmd-shift-p on a Mac). I changed my pico-configure to a different shortcut [...]

Can you solve this problem yet?