cpwood/Pico-Go #43: Connect connect to Pico

stephencross, 2021-02-26:
## What are the steps to reproduce this issue?
1.  Plug pico into USB porr

## What happens?

Never gets past:

Connecting to /dev/ttyACM0...

## What were you expecting to happen?

Expect to see:

Connecting to /dev/ttyACM0...
>>>

## Any logs, error output, etc?
*(If it’s long, please paste to https://gist.github.com and insert the link here)*

## Any other comments?

I've tried turning off auto_connect and setting the port manually.
Running Pop_OS 20.04

## Support info

Copy this from the `Pico-Go > Extra > Get support info` option in the command palette:

No available

cpwood, 2021-03-03:
Hi - sorry for the delay in replying. Could this be a permissions issue?

http://pico-go.net/docs/help/connect/#linux-users

cpwood, 2021-03-06:
Closing for now as there's no further action I can take, but feel free to re-open, if required.

closed 2021-03-06 by cpwood