cpwood/Pico-Go #47: Option to disable Pico Console

nvmker, 2021-03-04:
Hi,
Currently if Pico-Go extension enabled globally when I'm pressing ctrl+` to open terminal, VS Code opens Pico Console and I need to start default shell manually.
Could you please add an option to disable Pico Console or change behavior of starting it when open_on_start is set to false?

cpwood, 2021-03-06:
Sounds like a great idea! Yes, I'll work on this.

nvmker, 2021-03-31:
Since latest 1.55 release of VSCode added a possibility to change default terminal this request isn't more actual.

nvmker, 2021-04-01:
I reopen this issue because I still have an issue with Pico Console popping up, even if I set default shell to *sh.

hobbyhack, 2021-05-26:
The terminal should only open if it was open when the workspace was closed in vscode.  This is how the default terminal works. However, when this addon is enabled, it open every time no matter what.  It also replaces the default terminal in every project.