cpwood/Pico-Go #51: Pico Console UPDATE

kevindawson, 2021-03-06:
## What are the steps to reproduce this issue?
I have an example for you
```python
# pico-001.py

import random
import usys
from utime import sleep_ms, sleep
from machine import ADC, Pin

onboard_temp_sensor = ADC(4)
onboard_p25_led = Pin(25, Pin.OUT)
conversion_factor = 3.3 / (65535)
ON = True
OFF = False

def onboard_led(state):
    if state is True:
        onboard_p25_led.value(1)
        # led.high() additional for Pico
        sleep_ms(222)
    else:
        onboard_p25_led.value(0)
        # led.low() additional for Pico
        sleep_ms(181)

def onboard_temp():
    current_voltage = onboard_temp_sensor.read_u16() * conversion_factor
    return 27 - ((current_voltage - 0.706) / 0.001721)

def main():
    print("hello\n")

    for _ in range(random.randint(3, 5)):
        onboard_led(ON)
        # degres in unicode -> \u00b0 -> °
        print("Pico onboard temp is: %2.1f\u00b0C" % onboard_temp())
        onboard_led(OFF)
        sleep(1)

if __name__ == "__main__":
    main()
    print("\n\tSo long and Thanks for all the Fish!\n")
    usys.exit(0)

```

## What happens?
Pico Console updates on completion

## What were you expecting to happen?
Pico Console to update while script running

## Any logs, error output, etc?
n/a

## Any other comments?

## Support info

```
Pico-Go:      1.2.1
VS Code:      1.54.1
Electron:     11.3.0
Modules:      85
Node:         12.18.3
Platform:     linux
Architecture: x64
Board:        Raspberry Pi Pico with RP2040
Firmware:     v1.14 on 2021-02-24 (GNU 10.2.0 MinSizeRel)
```

cpwood, 2021-03-07:
Thanks for this! Can you confirm how you're executing it? Are you click the `Run` button _or_ transferring it to the board and calling it `main.py`, for example?

cpwood, 2021-03-07:
Think I've got this working a little better:

[image]

Needs a bit more testing yet though.

kevindawson, 2021-03-07:
Just ```Run``` Button

- nice gif above:)
- running as it did in previous versions

Cool

So long and Thanks for all the Fish!

kevindawson, 2021-03-07:
@cpwood 
re: Needs a bit more testing yet though.

I  have some time today to test if you want, just point the way

cpwood, 2021-03-07:
Might be a little bit late in the day, depending upon where you are in the world, but if you fancied having a play with version 1.2.2, I'd appreciate your feedback!

You can download it [here](https://cwood.blob.core.windows.net/temp/pico-go-1.2.2.vsix); I haven't uploaded it to the Marketplace just yet.

It's been a pretty tough fix, believe it or not. I got it working as per the GIF above fairly easily, but that then broke everything else (e.g. deleting files, uploading or downloading the project, etc). Hopefully cracked it now though.

I've also done quite a bit of work on reducing the CPU load of the extension.

kevindawson, 2021-03-07:
@cpwood 

beta 1.2.2

Nice One ;)

- CPU load of the extension - Looks like you killed the Hog - now 2% constant ;)
- able to Ctrl+C out of running code

have you ever considered displaying files on Pico with  ```print('uos.listdir :   ' + str(uos.listdir()))```

cpwood, 2021-03-07:
@kevindawson I've got something else up my sleeve for managing files 😉 Watch this space.

Thanks for the feedback; I'll upload it to the Marketplace tomorrow evening (UK time) assuming I don't find any gremlins before then.

kevindawson, 2021-03-07:
@cpwood 

Cool - is there any other time but GMT (primary time zone like yourself:)
So long and Thanks for all the Fish!

commit 8f055163 (2021-03-07): Fix #51
closed 2021-03-08 by cpwood