cpwood/Pico-Go #52: Reduce CPU load of Python terminal implementation

cpwood, 2021-03-07:
The new `terminalExec.py` has a tendency for high CPU usage. Need to:

1. remove the call to `lsock.setblocking(False)`
2. introduce a small delay in the never-ending `While` loop: `time.sleep(0.001)`

cpwood, 2021-03-07:
If anybody wants to try this out in the next 24 hours before I put it on the marketplace, please download 1.2.2 [here](https://cwood.blob.core.windows.net/temp/pico-go-1.2.2.vsix).

shanff, 2021-03-08:
Checked out quickly.  Dropped from 20% CPU with VSCode idle  on an i7-37770 to 7-8% with VSCode idle.  It's an improvement, but I'd rather see much less usage if it's not doing anything
Good work though.

cpwood, 2021-03-08:
I've done some more tweaks I'm seeing a 1% max CPU usage on my slightly aged laptop. Will release the changes today.

commit 629b245c (2021-03-07): Fix #52
commit e9af7d27 (2021-03-07): Fix #52
commit 457c5d4b (2021-03-08): Fix #52 - further refinements
closed 2021-03-08 by cpwood