cpwood/Pico-Go #54: Automate Stub Generation and Commits

charathram, 2021-03-08:
# Feature Request 🚀

## Is your feature request related to a problem? Please describe.
Right now, stub generation seems to be manual. Wouldn't it be better if it were automated? That way, users of Pico-Go would be guaranteed of having the most current stubs available to them at all times.

## Describe the solution you'd like
Automated stub generation would be ideal. Of course, the issue is that it would require the installation of a new version of the Pico SDK on a Pico board and the generation of the stub files. So, I'm not sure if it's feasible.

## Describe alternatives you've considered
None. Sorry!

## Teachability, Documentation, Adoption, Migration Strategy
To me, this feature would make Pico-Go become something I would adopt "in production" so that I'm not dependent on the maintainers to generate new stubs as each new version of the Pico Micropython port is released.

cpwood, 2021-03-08:
Hi @charathram! Thanks for the suggestion.

An important thing that's not immediately obvious about generated stubs is that they're.. well... a bit rubbish really. The stubs in Pico-Go are from my [Pico-Stub](https://github.com/cpwood/Pico-Stub) repo, and you can find _automatically_ generated stubs [here](https://github.com/cpwood/Pico-Stub/tree/main/generated).

I only use the generated stubs to spot what's changed between firmware versions. Beyond that, there's no documentation, no parameters on any of the functions, etc. Because of that I don't feel it's a very useful tool in itself; it gives you some sense of the classes and functions you can call, but not what they're for or which parameters you're expected to pass.

For that reason, the stubs for Pico-Go are written by hand and updated by me for every firmware release. As an example, compare [this automatically generated method](https://github.com/cpwood/Pico-Stub/blob/main/generated/machine.py#L43) with the [version that's been written by hand](https://github.com/cpwood/Pico-Stub/blob/main/frozen/machine.pyi#L703). As you can see, the latter is much richer in information and more useful to a developer.

I think the stubs from Pico-Stub are better than what's available in any other MicroPython implementation because they're maintained by people.

With that being the case, there's no way to automate things and I don't really want to lower the quality just for automation. As you rightly point out, it also requires hardware to be available all the time, automation around updating the firmware as new firmware becomes available.

I think it's probably better to have more people engage on the Pico-Stub project in terms of the maintenance so it's not just up to me to do it.

Hope this helps outline why this isn't a change I'm in a position to implement this particular request, but I do really appreciate your enthusiasm for Pico-Go.

closed 2021-03-08 by cpwood

charathram, 2021-03-08:
Thanks, @cpwood. Really appreciate the explanation. If there's some way in which I can engage with the project, please let me know. I'm a Micropython neophyte, but I really love Pico-Go and can see the promise of the project. Coding in VSCode is orders of magnitude better than in Thonny! :)