cpwood/Pico-Go #55: Potential feature - manage files on your Pico via FTP

cpwood, 2021-03-10:
I've prototyped an FTP server capability in Pico-Go; it would allow you to use FTP clients or FTP-based VS Code extensions to manage files on your Pico. 

Should it become a Pico-Go feature? I can't decide whether it's useful or I'm just adding feature bloat, so I'd love your feedback.

**[Place your vote here](https://fast-poll.com/poll/aaab8eb4#)** and provide any other comments below.

kevindawson, 2021-03-10:
@cpwood 

I am not able to vote Blindfolded
suggest you do an **unstable** release here (github) for us to look at.
- ...1.3.3.vsix

ps.
```text
Versioning scheme where in X.Y.Z

    "X" is the major release number. This only changes when there has been a significant break in the project.

    "Y" is an even number for stable versions, and is an *odd number for unstable version*s. For example, the 3.4.1 version is stable, but 3.5.1 is unstable.

    "Z" is used in "odd numbered" releases as an increment towards the "even numbered" release. After the stable release is made, it is then used for small bug fix "point releases."
```

cpwood, 2021-03-10:
@kevindawson the fundamental use-case here is file management of the board's filing system using a GUI.

I was trying to ascertain whether it was worth the effort of going down that path of doing _any_ kind of release at all, really. If _nobody's_ interested, there's no point me putting in the effort to even do that! 😄 

Where did you get that versioning info from..? Don't think it's from Pico-Go.. unless I'm forgetting things.

cpwood, 2021-03-10:
Here's a little demo of the prototype I've hacked together.

It shows me:

1. starting the FTP server
2. connecting using FileZilla (though any FTP client should work)
3. listing the contents of the board
4. downloading a file from the board to my desktop
5. copying a file from my desktop to the board
6. disconnecting and stopping the FTP server
7. listing the files on the board using Python and showing that `temp2.py` is now on the board.

[image]

kevindawson, 2021-03-11:
@cpwood 

1, the need to assist users with viewing file contents of a Pico YES 
```python
uos.listdir()
```
or a Tree view
or ls -l from **rshell**

I currently use rshell in vscode from my python vetrual console
[rshell on Pico forum](https://www.raspberrypi.org/forums/viewtopic.php?f=144&t=304992&hilit=pico+rshell&sid=622589ffb5971d509064dd29937d8a0d)

2, launching a GUI ftp client
- zilla thumbs up
- But is this part of "Pico-Go"
- I think it would better in Pico-Ftp or Ftp-Go But must be able to co-exist with Pico-Go, 

3, Where did you get that versioning info from...

>  cpwood commented on 9 Feb • [issue-15](https://github.com/cpwood/Pico-Go/issues/15)
```
OK, thanks for the feedback. FWIW, I've just uploaded 1.0.8-arm64 (which is a pre-release) but there's no actual change as far as Linux ARM64 is concerned (the change is macOS ARM64 which I can test and appears to be OK at first glance).
```

>   cpwood commented 29 days ago • [issue-15](https://github.com/cpwood/Pico-Go/issues/15)
```
Here's a download link for another beta - does this work any better for you? If so, it seems cross-compiling for ARM64 on an x64 machine is a bit dodgy (which is what I was attempting previously).
```
>  cpwood commented 4 days ago • [issue-51](https://github.com/cpwood/Pico-Go/issues/51)
```
You can download it here; I haven't uploaded it to the Marketplace just yet
```
In the above it is normal to use **unstable release** to avoid ambiguity

cpwood, 2021-03-13:
Final position on this is that the feature _will_ be included in v1.3.0.

In terms of the feedback received:

* versioning - I'd prefer to use SemVer, but acknowledge the point made and will make proper use of suffixed version numbers in future (e.g. `vX.Y.Z-alpha1`)
* other extension vs Pico-Go - given that only one application can be connected to the device via the serial port at a time, the idea of stopping Pico-Go to start up an additional FTP extension seems unnecessarily painful. The FTP capability _isn't_ turned on by default, so if people aren't interested in it, they can simply not start it.

kevindawson, 2021-03-13:
@cpwood 
cool looking forward to it

closed 2021-03-15 by cpwood