cpwood/Pico-Go #56: Raspberry Pi OS not able to configure Project

fidoriel, 2021-03-12:
## What are the steps to reproduce this issue?
1. Open folder with Python file
2. cmd+shift+p -> Configure Project

Same behaviour with alt+shift+cmd+i

## What happens?
error "pymakr.initialise failed"
Project is not configured

## What were you expecting to happen?
Project configuration successful.

## Any other comments?
I am using the latest 32 bit RPiOS with VSCode 1.5.2 and Pico-Go 1.2.2
On my Mac I am also using VSCode 1.5.2 and Pico-Go 1.2.2. There it is working fine.
It seems that on the Raspberry Pi the extension is not loaded at the status bar like on macOS

I know there have been similar issues but they have been closed. e.g. Issue #38, but this is a different platform ans OS.

cpwood, 2021-03-12:
Very odd, since 32-bit ARM isn't supported; but you should've been notified of this upon startup. 😕 Did that not happen?

I would like to add ARM32, but it's not there just yet.

fidoriel, 2021-03-12:
I am sorry. You are completely right. I was so happy to have this Extension that I did not notice although it is a system requirement.
No, I did not receive a "Not supported" error in vscode. 
I will wait patient for support.

closed 2021-03-13 by cpwood

aannenko, 2021-04-11:
I have ARM64 version of Raspberry PI OS and the extension (1.3.1) gives back "command 'picogo.initialise' not found".
Any thoughts?

Other info:
- Raspberry PI 3b+
- `uname -a` output: `Linux raspberry 5.10.17-v8+ #1403 SMP PREEMPT Mon Feb 22 11:37:54 GMT 2021 aarch64 GNU/Linux`
- VS Code 1.55.1 on Windows,
- Connected to Raspberry PI OS using the "Remote - SSH" extension
- Pico connected, latest [.uf2](https://www.raspberrypi.org/documentation/rp2040/getting-started/static/f70cc2e37832cde5a107f6f2af06b4bc/rp2-pico-20210205-unstable-v1.14-8-g1f800cac3.uf2) flashed.