cpwood/Pico-Go #57: Unable to start terminal | terminated with exit code 1.

vondelden, 2021-03-12:
## What are the steps to reproduce this issue?
1. Install Visual Studio Code
2. Install Pico- Go extension
3. Restart Visual Studio Code

## What happens?
Everything loads normally except for pico go terminal.
https://user-images.githubusercontent.com/4185263/110977503-88cb7180-8362-11eb-9e0f-4f2fb5b6ae5b.mp4

Error code:
The terminal process "py.exe
'c:\Users\computername\.vscode\extensions\chriswood.pico-go-1.2.2\terminalExec.py', '1730'" terminated with exit code: 1.

## What were you expecting to happen?

## Any logs, error output, etc?
*(If it’s long, please paste to https://gist.github.com and insert the link here)*

The terminal process "py.exe 'c:\Users\computername\.vscode\extensions\chriswood.pico-go-1.2.2\terminalExec.py', '1730'" terminated with exit code: 1.
onDidChangeNotification @ notificationsAlerts.ts:40
console.ts:137 [Extension Host] [warning] Term | Error while connecting to term
console.ts:137 [Extension Host] [warning] Term | Error: connect ECONNREFUSED 127.0.0.1:1730
console.ts:137 [Extension Host] [warning] Term | Term connection closed
console.ts:137 [Extension Host] [warning] Term | true
console.ts:137 [Extension Host] [warning] Term | Error while connecting to term
console.ts:137 [Extension Host] [warning] Term | Error: connect ECONNREFUSED 127.0.0.1:1730

## Any other comments?

I've tried uninstalling VSC and installing older versions from pico-go from the version repository.

## Support info

Copy this from the `Pico-Go > Extra > Get support info` option in the command palette:

```
Sorry nothing happens if I click get support info.
```

MrBambix, 2021-03-12:
I am having the exact same problem platform windows 10 64b.

cpwood, 2021-03-12:
Hi folks - what happens if you run the command in the error output from a command line - e.g., from the above:

```
py.exe 'c:\Users\computername.vscode\extensions\chriswood.pico-go-1.2.2\terminalExec.py'
```

Can you paste any error output?

vondelden, 2021-03-12:
File "C:\Users\computername\.vscode\extensions\chriswood.pico-go-1.2.2\terminalExec.py", line 84
    def boardInput(data: bytes):
                       ^
SyntaxError: invalid syntax

cpwood, 2021-03-12:
Thanks, @vondelden! At first glance, that doesn't really look like a syntax error to me (i.e. I can't see anything wrong with that line of code.. so I'm a bit confused!

If you happen to be a Python programmer and can spot a problem when you open `terminalExec.py` and look near that line, can you let me know what you see?

Many thanks! I'll see if I can figure something out over the weekend when I've got a bit more time, too.

vondelden, 2021-03-12:
Thank you for creating a much needed program. I will have a look tho I'm a beginner in python so I won't be much help. Many thanks for taking the time.

cpwood, 2021-03-12:
Thanks, @vondelden!

Could I just ask you to do one more thing? What version number do you get when you run:

```
py.exe -V
```

It _should_ be 3.x.y.

vondelden, 2021-03-12:
Python 3.9.1

ags131, 2021-03-12:
I'm having the same issue when both python 2 and 3 are installed, for me the fix was to edit the first line from 
```python
#!/usr/bin/python
```
to
```python
#!/usr/bin/python3
```

cpwood, 2021-03-12:
Many thanks, @ags131 ! Does this work for anybody else?

If it does, I'll get an update rolled out tomorrow.

MrBambix, 2021-03-13:
Hi

I will test this today and let you know.

Greetz
Leo

Verstuurd vanaf mijn iPhone

MrBambix, 2021-03-13:
Sorry to say that changing the first line of terminalExec.py to #!/usr/bin/python3 did not help in my case. However getting totally rid of the line did work under vscode on windows 10. I have python3 on the path so probably this line is not needed i guess as it sets a specific interpreter just for this file.

regards
Leo
ps great plugin, thank you !

vondelden, 2021-03-13:
@cpwood I have got a working pico console now. I also had python 2 installed. I should've tested changing it to #!usr/bin/python3 first but I went too fast and and removed python 2 from my machine. At least it works now. Great extension!

closed 2021-03-13 by cpwood

cpwood, 2021-03-13:
v1.2.3 will be available in the Marketplace any moment. Thanks, everyone!

commit ba808d99 (2021-03-13): Fix #57