cpwood/Pico-Go #65: Extension installed, but somehow not

Chr1s70ph, 2021-03-17:
## What are the steps to reproduce this issue?
1. Install the extension
2. Try to run any commands

## What happens?
I installed the extension right in VS Code, and tried to run the commands, after it was installed. I even restarted the IDE.
As soon as I issue any Command, I get an Error, saying that "picogo.initialise" (e.g. for Configure Project) has not been found.

## What were you expecting to happen?
I expect, that once I issue a command, it executes its function

## Any logs, error output, etc?
Command 'Pico-Go > Configure project' resulted in an error (command 'picogo.initialise' not found)

## Any other comments?
I have looked in the Settings under Extensions, and there the Extension does not even show up. I have restarted the IDE multiple times, un- and reinstalled the extension multiple times, even restarted my PC more than once, still no change in outcome

## Support info

Copy this from the `Pico-Go > Help > Get support info` option in the command palette:

```
Command 'Pico-Go > Configure project' resulted in an error (command 'picogo.initialise' not found)
```

cpwood, 2021-03-17:
Hi there; sounds very odd! 

To check a few basics: can you confirm that:

* you've got Python 3 installed (check using `py -V` from Command Prompt);
* you're on a 64-bit version of Windows 10;
* whether there's anything attributable to Pico-Go when you load the Developer tools from the Help menu and switch to the Console tab?

Thanks in advance!

Chris

Chr1s70ph, 2021-03-18:
Hi, thanks for the reply. 
Id did what you asked me to:
> To check a few basics: can you confirm that:
> 
>     * you've got Python 3 installed (check using `py -V` from Command Prompt);

and weirdly, even tho I had pyhton installed, it kept telling me that it wasn't.
I installed python previously over the Windows store (don't ask why initially, I just kept rolling with it). Now on the other hand, I installed it from the official website, and would you look at that, it works now...
Thanks for your patience and help 😄

closed 2021-03-18 by Chr1s70ph