cpwood/Pico-Go #66: Endless crash loop when no Pico is attached

epmoyer, 2021-03-18:
## What are the steps to reproduce this issue?
1. Install Pico-Go in MacOS with no Pico attached.

## What happens?
VSCode goes into an infinite loop repeatedly trying to open a Pico Console

## What were you expecting to happen?
- Should fail once and stop somehow, presumably with a way to retry later after attaching a Pico8.
- It seems "lucky" at the moment that I am able to retain enough control of VSCode to hit the "Uninstall" button to stop the extension. It is probably good practice with a VSCode extension like this to insert some kind of sleep/delay on a failure. As it stands now, the crash loop very nearly brings the app to a standstill.  It is not possible to navigate menus, and I only managed to uninstall because the Pico-Go extension page happened to be visible and my "Uninstall" button click managed to be recognized by the hammered event loop.

## Any logs, error output, etc?
[image]
[image]

## Any other comments?
I don't yet have a Pico in hand (it arrives later today), so I can't yet test the "working" case.  Presumably it would successfully open a terminal if I had a Pico attached.

## Support info
Can't get control long enough to gather this.
System is:
- MacOS 11.2.1
- VSCode Version: 1.51.0, Commit: fcac248b077b55bae4ba5bab613fd6e9156c2f0c

cpwood, 2021-03-18:
Thanks for this! One immediate comment is that you're using a very old version of Pico-Go there. Current version is 1.3.

epmoyer, 2021-03-18:
Thanks Chris!
I'm not sure why that error message says "pico-go-1.0.4".  The Extensions browser lists 1.3.0, and I'm just clicking "Install" on that. Also I've never installed it before today, so It seems unlikely that my system would have had an old version sitting around prior to today's install.  As soon as my device arrives I'll do some experimentation with the device both attached and removed.

Thanks again for the quick response; cheers!

[image]

epmoyer, 2021-03-18:
[image]
I neglected to mention above that both python3 and node are in my path.  Mail truck just pulled up, so finger crossed!

cpwood, 2021-03-18:
I can't explain that, sorry. It's definitely _not_ 1.3 though as `terminalExec.js` doesn't exist anymore from 1.2 onward. 

I'd probably recommend uninstalling, clearing out anything related to Pico-Go from your `.vscode/extensions` folder and then reinstalling again.

If it still produces an error, try running the failed command at a terminal window and see whether anything useful appears in the error output.

epmoyer, 2021-03-18:
Okay, for some reason VSCode was stuck at an older version and not updating.  I forced the update and now VSCode is at Version: 1.54.3, and it correctly installs Pico-Go 1.3.0.

1.3.0 is failing with a similar message, but this time terminalExec.py is failing to launch.  I now have a Pico attached, and it will connect to Thonny, so in general the device side is working so I am guessing it is not that.

I can't find any logs for what problem terminalExec.py is encountering.  Any idea where to look?
[image]

epmoyer, 2021-03-18:
**Solved**

I hard coded my python3 path in `chriswood.pico-go-1.3.0/lib/main/terminal.js` and that solved the problem.  I run Anaconda, and I suspect that somehow python3 is not in the path at the time the extension tries to load.   I will try to find a better long-term solution, but the problem appears to be mine and not yours in any case :) 
[image]

Now that I am running 1.3.0, and have the python3 path resolved, VSCode is not stuck in a restart loop even if the Pico is disconnected on launch (I just see the message below, which seems normal), so this issue is closed.

[image]

Thanks for being so responsive Chris.  Cheers!

closed 2021-03-18 by epmoyer