cpwood/Pico-Go #68: Pico-Go does not support Anaconda

Ethycs, 2021-03-21:
## What are the steps to reproduce this issue?
1. Use Miniconda or Conda to manage python environment
2. Install Pico-go
3. Attempt to use Pico-go

## What happens?
Pico-go does not recognize Anaconda or miniconda managed python

## What were you expecting to happen?
Pico-go should find python or at least provide a way to set the path

## Any logs, error output, etc?
*(If it’s long, please paste to https://gist.github.com and insert the link here)*
Python3 is not detected on this machine so Pico-Go cannot work. Ensure it is in your PATH.

## Any other comments?

## Support info

Copy this from the `Pico-Go > Help > Get support info` option in the command palette:

```
Command 'Pico-Go > Help > Get support info' resulted in an error (command 'picogo.extra.getFullVersion' not found)
```

cpwood, 2021-03-21:
Thanks for this! 

Since I'm not familiar with Conda's virtual environments, can you confirm if Pico-Go still doesn't work if you've activated an environment first? From my reading, I would've  expected the Python interpreter is  added to your PATH in doing so. 

FWIW, we need Python to run a system-level script that acts as a proxy between the extension and the VS Code terminal window. So I don't think it's a scenario where you'd prefer one version of Python over another at project level. We just need something at system level for the extension to run.

Being able to configure the Python path manually sounds like a good enhancement to include.

Ethycs, 2021-03-21:
Can confirm, a basic work around for another plugin is to start a shell with an Anaconda environment and then run vscode using the command `code` in Windows. 

[image]

Ethycs, 2021-03-21:
> FWIW, we need Python to run a system-level script that acts as a proxy between the extension and the VS Code terminal window. So I don't think it's a scenario where you'd prefer one version of Python over another at project level. We just need something at system level for the extension to run.

Anaconda does not make python available on my system path by default. I'm actually quite happy with this. This may seem nonsensical but it has been very effective in preventing Python packages from breaking my various environments.  Python is not added to the system path by default, AFAIK but the user path

cpwood, 2021-03-27:
The ability to set a manual `python_path` value in the Global Settings will be included in v1.3.1.

closed 2021-03-28 by cpwood