cpwood/Pico-Go #69: Request more useful error message

chuckuntulis, 2021-03-25:
# Feature Request 🚀

I have Pico-go 1.3.0  installed in VSCode. Pico-go>Run current file generates a "command 'picogo.run' not found" error. It would be useful to suggest some reason(s) why this might happen and what to do to solve the problem because I am stuck at this point.

Similarly Command 'Pico-Go > Help > Getting started' resulted in an error

When I open the View>Command Palette and select any of the Pico-go commands. they all fail.

If there is something else necessary to do beside the install of Pico-go, it would be useful to describe this.

cpwood, 2021-03-27:
Thanks! This will be fixed in v1.3.1.

chuckuntulis, 2021-03-27:
Thanks. Why would it work on one system and not the other? I get the
same error message for the CircuitPython extension which leads me to
believe that I did something wrong? Is there some place that I can look or
try. I am willing to do the testing or detective work.

On another issue - in your flash example, when I try to import the machine
library it complains that it can't find it and consequently can't get the
blink example to work. If I use the code from "Colin's Lab" Adafruit Mu
example I can get the LED blinking using different libraries.

I am still confused about where to find the missing "machine" library and
where VSCode is looking for it.

Thanks.

cpwood, 2021-03-27:
Have your seen and worked through the Quick Start guide? I get the sense you're not quite set up properly. 

http://pico-go.net/docs/start/quick/

My best guess is that you've not initialised the project.

chuckuntulis, 2021-03-27:
I thought I had, but I will uninstall Pico-go and follow the directions
precisely.
Thanks for the suggestion.

chuckuntulis, 2021-03-27:
Following the directions:

I have Python 3.8.8 installed on the system.

There are three Python options for VSCode: Python for VSCode,
PythonExtension and ms-python.python. I have the later installed. Is that
the right one or does it not matter?

On to the next steps.

cpwood, 2021-03-27:
Take a look at `.vscode/extensions.json` once you've initialised the project:

```json
{
    "recommendations": [
        "ms-python.python",
        "visualstudioexptteam.vscodeintellicode",
        "ms-python.vscode-pylance"
    ]
}
```

You'll want the lot.

chuckuntulis, 2021-03-27:
I can not copy text from your Getting Started page. I can copy text from
other pages? Is there a reason for that?

Pico-go 1.3.0 is installed

There is no Pico-go status bar at the bottom of VSCode as there is on the
other system?

I can find the Pico-go Create Project in the list of Pico-go commands, but
when I select it, I get "Command 'Pico-Go > Configure project' resulted in
an error (command 'picogo.initialise' not found)"

What next? I will keep poking around.

chuckuntulis, 2021-03-27:
Since the project has not been created by Pico-go, this file does not exist
in the directory that I created as per the directions.

cpwood, 2021-03-27:
I'm a little lost on where you're up to, if I'm honest!

Can you confirm you're definitely running one of the supported platforms: 

* Windows x64
* Linux x64 or arm64 (**not** arm32)
* macOS x64 or arm64?

If you're getting errors running *any* Pico-Go command, it suggests Python might be installed but isn't in your PATH. Try typing one of the following:

Linux / Mac Terminal:

```
python3 -V
```

Windows Command Prompt:

```
py -V
```

If you only get errors, Python isn't installed within your path and needs to be.

cpwood, 2021-03-27:
PS - thanks for the heads-up on being unable to copy from the website. Never noticed that before! I'll try and turn whatever's trying to be clever off!

chuckuntulis, 2021-03-27:
at the command prompt, if I run python -V, I see Python 3.8.8. If I run py
-V the error is that py is not recognised.

chuckuntulis, 2021-03-27:
OK

cpwood, 2021-03-27:
Can you confirm which OS you're using? When Pico-Go starts up, it runs a Python script which acts as a kind of proxy between the extension and the terminal window. On Windows, it attempts to launch this with `py`, but on macOS it attempts to use `python3`.

If you're on Windows and `py` returns an error, that'll be why it's not working. Maybe I should get it to try both..!

Version 1.3.1 (out tomorrow) will allow you to set the path to Python manually, if needs be.

chuckuntulis, 2021-03-27:
Sorry. Windows 10 Home 64bit version 20H2

chuckuntulis, 2021-03-27:
python
>>> import sys
>>> print(sys.path)
['', 'C:\\Program
Files\\WindowsApps\\PythonSoftwareFoundation.Python.3.8_3.8.2288.0_x64__qbz5n2kfra8p0\\python38.zip',
'C:\\Program
Files\\WindowsApps\\PythonSoftwareFoundation.Python.3.8_3.8.2288.0_x64__qbz5n2kfra8p0\\DLLs',
'C:\\Program
Files\\WindowsApps\\PythonSoftwareFoundation.Python.3.8_3.8.2288.0_x64__qbz5n2kfra8p0\\lib',
'C:\\Users\\Charles\\AppData\\Local\\Microsoft\\WindowsApps\\PythonSoftwareFoundation.Python.3.8_qbz5n2kfra8p0',
'C:\\Users\\Charles\\AppData\\Local\\Packages\\PythonSoftwareFoundation.Python.3.8_qbz5n2kfra8p0\\LocalCache\\local-packages\\Python38\\site-packages',
'C:\\Users\\Charles\\AppData\\Local\\Packages\\PythonSoftwareFoundation.Python.3.8_qbz5n2kfra8p0\\LocalCache\\local-packages\\Python38\\site-packages\\win32',
'C:\\Users\\Charles\\AppData\\Local\\Packages\\PythonSoftwareFoundation.Python.3.8_qbz5n2kfra8p0\\LocalCache\\local-packages\\Python38\\site-packages\\win32\\lib',
'C:\\Users\\Charles\\AppData\\Local\\Packages\\PythonSoftwareFoundation.Python.3.8_qbz5n2kfra8p0\\LocalCache\\local-packages\\Python38\\site-packages\\Pythonwin',
'C:\\Program
Files\\WindowsApps\\PythonSoftwareFoundation.Python.3.8_3.8.2288.0_x64__qbz5n2kfra8p0',
'C:\\Program
Files\\WindowsApps\\PythonSoftwareFoundation.Python.3.8_3.8.2288.0_x64__qbz5n2kfra8p0\\lib\\site-packages']

cpwood, 2021-03-27:
Installed from the Windows Store, by the looks of it? Mine's installed from the Python web site, so that's probably the difference.

OK - leave it with me. I'll get it to try both `py` and `python3` on Windows before failing.

cpwood, 2021-03-27:
From a Pico-Go perspective, tomorrow's release will fix things for you. As for any broader issues, an alias might work, yes; though it's not anything I've run into before, myself.

chuckuntulis, 2021-03-27:
If I find out anything, I will let you know.

chuckuntulis, 2021-03-28:
I  am still trying to minimize my confusion and build a model of the
various artifacts of Python extension, Pico-go extension, CircuitPython
extension, Mu, VisualStudioCode  and the libraries being used in examples -
where they are physically located and how they are logically referenced.

I looked at some threads about aliases and I tried "doskey py=python" but
that didn't work. i will keep researching how to set up an alias for py to
Python.

I looked at the Mu directory and found that it included a Python
Application from 7/6/2019 and a python36.zip from 6/27/20 so I assume
these are connected in some way.

I have found the pico-go files on my system in
C:\Users\Charles\.vscode\extensions\chriswood.pico-go-1.2.0\node_modules\electron\dist\Electron.app\Contents\Resources
but will not be able to find much information until I can unpack the ASAR
file (if possible) and actually look at the code. Is that possible and if
so how?

The example file is:
import time
import board
import digitalio
led=digitalio.DigitalInOut(board.LED)
led.switch_to_output()
while True:
    led.value=True
    time.sleep(0.1)
    led.value=False
    time.sleep(0.1)

Will your fix tomorrow resolve this issue?

Thanks
Chuck

chuckuntulis, 2021-03-28:
Hi Chris,
I just tried py at the command prompt in VSCode and in the powershell and
they invoked Python on one system and not on the other system?

I have reported the BUG to the author of CircuitPython extension because
those commands also fail.

chuckuntulis, 2021-03-28:
I was going to report this BUG to VSCode, when entered Pico-go, I received "
The issue reporter is unable to create issues for this extension, as it
does not specify a URL for reporting issues. Please check the marketplace
page of this extension to see if other instructions are available.

chuckuntulis, 2021-03-28:
I was able to get the CircuitPython commands to work after I plugged a RPi
Pico into the system - of course that makes a lot of sense now.

I will slowly figure out all this stuff out.

closed 2021-03-28 by cpwood

chuckuntulis, 2021-03-28:
> Closed #69 <https://github.com/cpwood/Pico-Go/issues/69> via be8c907

I am afraid that the problem still exists. I have loaded an older version
and then the latest version 1.3.1, but the Pico-go prompts are not in the
lower border of VSCode and when I try any of the View>Command
Palette>Pico-go commands, I still get the same error messages.

I still can't run py on the second system. I will be working on that.

What next?

chuckuntulis, 2021-03-28:
On the failing system:

In VSCode Terminal I see
PS E:\Stuff\Python stuff\PicoProject> python
Python 3.8.8 (tags/v3.8.8:024d805, Feb 19 2021, 13:18:16) [MSC v.1928 64
bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>>

Powershell
PS C:\Users\Charles> python
Python 3.8.8 (tags/v3.8.8:024d805, Feb 19 2021, 13:18:16) [MSC v.1928 64
bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>>

Is there anything that I can look at on my system that I can look at to ret
to determine why it is running on one system and not the other?

chuckuntulis, 2021-03-29:
I installed the latest version of Python and ended up with a py.exe in
C:\Windows. Now when I do a py in VSCode it fires up Python and it also
works when invoking it from the Power Shell.

Pico-go commands still generate error messages that are not helpful for
identifying what the error is or even what part o Pico-go code is being
executed when the error occurs.

Can you send me (or point me to)  the code so I can plant some error prints
or breakpoints?

Thanks
Chuck

cpwood, 2021-03-29:
Based on what you wrote above, I'd expect the _current_ version of the extension to be here on your computer:

`C:\Users\Charles\.vscode\extensions\chriswood.pico-go-1.3.1\lib`

If something's going wrong, it will most likely be in `activator.js`.

chuckuntulis, 2021-03-29:
Thanks. I will take a look.

chuckuntulis, 2021-03-29:
Thanks. I have found the activator file but I don't have much experience
with Javascript.  I will just keep plodding along.

chuckuntulis, 2021-03-30:
Hi Chris,

I have looked at the code and nothing jumps out at me. I will keep looking.
How different is your Activator script than the Activator for any other
extension. Is there a lot of boiler plate involved in developing an
extension? I will be looking at
https://code.visualstudio.com/api/get-started/your-first-extension.

Is it possible to put print statements into your activator script in
C:\Users\Charles\.vscode\extensions\chriswood.pico-go-1.3.1\libactivator.
If so, where would the printout appear - in OUTPUT, DEBUG CONSOLE or
TERMINAL?

Thanks

chuckuntulis, 2021-03-30:
Hi Chris,

 I have searched, but I have not found the actual Pico-pins.png and where
it's path is defined?

Thanks

chuckuntulis, 2021-03-30:
Found it.

chuckuntulis, 2021-03-30:
If I can figure out how to add print statements to activator.js , then I
will be able to see if it fails before or after constructing the imageURL
in _getPinMapHTML and then work from there.

chuckuntulis, 2021-03-31:
Hi Chris,

I am sorry to pester you, but under the assumption that the Activator.js
file is executed when the commands are called or when Pico-go is installed,
I put the following code in Activator:
    value: async function activate(context) {
      var _this = this;
      var sw = new _settingsWrapper2.default();
      await sw.initialize(context.workspaceState);

      await this._checkPythonVersion(sw);
      vscode.window.showErrorMessage('activate start');

but I never see the activate start message.

Even if I delete the activator.js I still get the Command 'Pico-Go > List
all Commands' resulted in an error (command 'picogo.listCommands' not found)
error message so it looks like Pico-go is never called. I have installed
and refreshed an earlier version of Pico-go with no changes and it fails in
the same way. I go back to 1.3.1 and refresh and I get the same message
even though Activator.js is not present in the lib directory. Is the code
being cached some place?

I selected uninstall, but the code is still in the extensions directory?

I renamed the directory and I can see that chriswood.pico-go-1.3.1 is not
in the extensions directory. I then uninstalled 1.3.1 but the
Pico-go commands are still available. I am confused.

Try as I might, I can not get a clean slate to test, and I can't get test
messages to be executed so I can not figure out why Pico-go is not working
like it is on the other machine.

Since I can't isolate and test whether it is a Pyhon related issue, I have
less confidence that your fix changed Pico-go's behaviour.

I also put a test message in the code that is supposed to show the pin map
and it does not appear to rely on Python (all the Python and py paths seem
to be set up) in any way but it does not work but shows the same generic
error message..

Any hints would be helpful.

commit be8c907b (2021-03-28): Fix #69