cpwood/Pico-Go #70: Ability to Select a different COM port when more than one Pico is connected

Idirianuk, 2021-03-26:
I think that the title says it all.

I have a number of Picos. At times it would be really helpful to switch between Picos without having to unplug and plug different ones when switching between projects.

Rather than manually and globally set a COM port to use, it would be brilliant if it was possible to select, one at a time, an alternative COM port to use instead of the default, detected one.

cpwood, 2021-03-27:
Interesting! Suspect that this will either be a _really_ easy change or a _really_ complicated one with no middle ground! Will investigate when I'm in front of my computer. 

In terms of the use case, is it a case of one Pico per project, or are you swapping Picos within a project?

Idirianuk, 2021-03-27:
Thanks for your response. The use case is having multiple Picos plugged in doing different things and switching projects; the desire being to be able to change vs.code sessions and work on a different project without needing to unplug/replug Picos. Some of the projects involve things running for a while and gathering data. It'd be excellent to be able to work on something else whilet another is doing its job.

I'm quite new to all of this, so unsure of what may be involved (though I have an inkling, not backed up by skill!). Former games programmer here, not done ANY coding for over 20 years and having fun with Pis and Picos and Python. It's more difficult than 68000 :P

Thanks. :)

cpwood, 2021-03-27:
@Idirianuk - any chance I could get you to try out [this beta version](https://cwood.blob.core.windows.net/temp/pico-go-1.3.1-beta1.vsix)?

You'll also need to change the **Global Settings** `auto_connect` value to be `false` and add the following to `pico-go.json` for your **Project Settings**:

```json
"manual_com_device": "COM4"
```

Each project can have its own `manual_com_device` value.

Unfortunately, I can't test this myself as I'm not lucky enough to have more than one Pico! Can you let me know if it does the trick?

If so, I'll get it released tomorrow (Sunday), hopefully.

Many thanks!

Idirianuk, 2021-03-27:
Hi, Chris. Thanks for you efforts.

For the "default" COM port (in my case, COM4 which seems to adopt primacy), all is well, and it's business as usual (with a modified global and local .json file).

With two other boards connected (to COM3 and COM7, here), and with the settings shown below, Pico-go attempts to establish a connection to the correct port (or at least is displays the correct port specified in the local pico-go.json), but appears to fail to establish a connection to said COM port.

When I manually click on the "X Pico Disconnected" button at vs.code's footer the following error is generated:

```
Connection attempt cancelled
> Failed to connect (Error: Writing to COM port (GetOverlappedResult): Operation aborted). Click the "Pico Disconnected" button to try again.
Connecting to COM3...
```
There's a further popup message at the bottom right with the message:

```
Running the contributed command: 'picogo.toggleConnect' failed.
```

Here's my local folder pico-go.json
```
{
    "sync_folder": "",
    "open_on_start": true,
    "manual_com_device": "COM3"
}
```
Here are my global settings (modified)

```
{
	"sync_folder": "",
	"open_on_start": false,
	"safe_boot_on_upload": false,
	"statusbar_buttons": null,
	"py_ignore": [
		"pico-go.json",
		".vscode",
		".gitignore",
		".git",
		"project.pico-go",
		"env",
		"venv"
	],
	"fast_upload": null,
	"sync_file_types": "py,txt,log,json,xml,html,js,css,mpy",
	"ctrl_c_on_connect": false,
	"sync_all_file_types": false,
	"auto_connect": false,
	"manual_com_device": null,
	"autoconnect_comport_manufacturers": [
		"MicroPython",
		"Microsoft"
	],
	"ftp_password": "pico"
}
```

I hope that's helpful, and once again, many thanks for your efforts. Pico-go is a superb helper in vs.code, exactly as it stands. Of course, the feature request is just that, a request, so no worries if it's too time-consuming or remains in the blue of the sky! :)

cpwood, 2021-03-27:
Hmm, think this is probably going to be a bit tricky to resolve, unfortunately - at least without another Pico.

Out of curiosity, if you just run just a single VS Code project at a time - each with a different Pico board on a different COM port - does that work OK? i.e. is it only when you have multiple projects open at the same time that it's unhappy?

cpwood, 2021-03-28:
Going to close this ticket for now on the assumption that I've at least added value around allowing different projects to use different COM ports, even if they can't be loaded concurrently.

closed 2021-03-28 by cpwood

Idirianuk, 2021-03-28:
Just the one project open at a time. 😀

Idirianuk, 2021-03-28:
Not sure what's changed, but this all is working now. Fabulous!