cpwood/Pico-Go #71: Unable to interact with Pico if main.py has infinite loop

thompsonalt, 2021-03-27:
## What are the steps to reproduce this issue?
1. Create simple forever loop in `main.py`. For example:
```
from machine import Pin
import utime

led = Pin(25, Pin.OUT)

while True:
    led.toggle()
    utime.sleep(2)
```
2. Upload it to the Pico
3. Attempt to access repl or run other scripts 

## What happens?
Pico console says `Connecting to COM4...` but never gives the `>>>` prompt. Attempting to run another script also fails with the only output being `MPY`.

## What were you expecting to happen?
I would expect to be able to connect to a Pico with a `main.py` that has an infinite loop. The only way I can get it out of this state from Pico-go is to make a blank `main.py` script and upload it. Then the repl comes back and everything works again.

## Any other comments?
Connecting with rshell works more as I would expect. The `main.py` program exits and then I have access to the shell. 

Disclaimer, I only just started experimenting with the Pico, so I definitely could be doing something wrong or misunderstanding how things are supposed to work. Thanks so much for providing this extension!

## Support info

```
Pico-Go:      1.3.0
VS Code:      1.54.3
Electron:     11.3.0
Modules:      85
Node:         12.18.3
Platform:     win32
Architecture: x64
Board:        Raspberry Pi Pico with RP2040
Firmware:     v1.14 on 2021-02-05 (GNU 9.3.0 MinSizeRel)
```

cpwood, 2021-03-27:
Hi there! Does clicking the reset button or typing Ctrl+C work? That's how I'd expect it to work, though I'll confess I've not checked it for a while!

closed 2021-03-27 by cpwood

thompsonalt, 2021-03-27:
Ha! Of course Ctrl+C works. Should have gotten some sleep before submitting a bug report. Thanks!

GregWoods, 2021-08-02:
This caught me out today. After moving my code from a src folder (and always using pico-go "Run") to the root of the pico, I was convinced i had somehow blown the board when it appeared unresponsive to connections. Eventually I hit Ctrl-C !!