cpwood/Pico-Go #72: Unable to use PicoGo - Command palette commands not found, python3 not found

elschopi, 2021-03-27:
Not so much a bug I guess, but more a call for guidance and/or enlightenment 😅 
## What are the steps to reproduce this issue?
1. install PicoGo extension from VSCode marketplace
2. reload / restart VSCode
3. try to use anything PicoGo related

## What happens?
when e.g. trying to get support info via the command palette command, I get the error message that the command could not be found.
As soon as VSCode starts, I get a error message telling me that python3 is not found and to ensure it is in my PATH.

## What were you expecting to happen?
extension to work

## Any logs, error output, etc?
*(If it’s long, please paste to https://gist.github.com and insert the link here)*

## Any other comments?
On my work PC (Win10 64 bit), I have installed Python 3.9.2 (enabled "add to PATH" option) and VS Code Insiders. Installed Pico Go along with Python extension. 
There, Pico Go works flawless.

On my home PC (Win 10 64bit), I have installed Python 3.9.2 (enabled "add to PATH" option) and VS Code Insiders. Installed Pico Go along with Python extension. 
Here, it does not work. I get the above mentioned error messages. I can't seem to be able to find the differences or what I did wrong.

## Support info
Error mesage in VS Code on startup: 
Python3 is not detected on this machine so Pico-Go cannot work. Ensure it is in your PATH.

Copy this from the `Pico-Go > Help > Get support info` option in the command palette:

```
Command 'Pico-Go > Help > Get support info' resulted in an error (command 'picogo.extra.getFullVersion' not found)
```

cpwood, 2021-03-27:
Hi there - firstly, I suppose it's worth checking that Pico-Go is correct in saying Python isn't in your path, just in case you really have hit upon a bug. If you go to a Command Prompt window and type `py -V`, you should get the Python version number appearing.

If that doesn't appear, the extension is right in saying that Python isn't in your path. Method 2 here should help:

https://datatofish.com/add-python-to-windows-path/

Hope that helps!

cpwood, 2021-03-27:
A better alert (instead of the command not found one) will be included in v1.3.1, too.

elschopi, 2021-03-27:
Hi,
thanks for the quick answer! "py -V" brings no result in my ordinary command prompt, "python --version" brings me the installed python version. I assumed that this means that it's in my PATH, please correct me if I'm wrong.

closed 2021-03-28 by cpwood

lorand44, 2021-04-22:
I know it's closed, but I want to answer @elschopi . I had the same issue. This page helped me realise that I didn't have python installed correctly. I accidentally deleted some folders from the common python folder and that resulted `py -V` giving me a `Can't find default python'` error. I reinstalled python and suddenly everything works. :)