cpwood/Pico-Go #75: Problem connecting to the pico

adarmaori, 2021-04-02:
## What are the steps to reproduce this issue?
1. disable auto_connect in the global settings
2. set manual_com_device to the port the pico is connected to

## What happens?
The output in the terminal is `Address not configured. Please go to the settings to configure a valid address or comport`

This is the settings JSON after I changed it:
```{
    "sync_folder": "",
    "open_on_start": false,
    "safe_boot_on_upload": false,
    "statusbar_buttons": null,
    "py_ignore": ["pico-go.json", ".vscode", ".gitignore", ".git", "project.pico-go", "env", "venv"],
    "fast_upload": null,
    "sync_file_types": "py,txt,log,json,xml,html,js,css,mpy",
    "ctrl_c_on_connect": false,
    "sync_all_file_types": false,
    "auto_connect": false,
    "manual_com_device": "COM5",
    "autoconnect_comport_manufacturers": ["MicroPython", "Microsoft"],
    "ftp_password": "pico"
}```
Did I do something wrong or is this a bug?

cpwood, 2021-04-02:
Nothing's jumping out as being wrong to me... Out of curiosity, if you put the `auto_connect` and `manual_com_device` values into your project's `pico-go.json`, does that change things at all?

adarmaori, 2021-04-02:
It ignores the change and tries again to connect to the wrong port

shalva97, 2021-06-11:
any updates on this? my port is also COM5 but it still tries to use COM3

shalva97, 2021-06-11:
my bad, did not set auto_connect to false. its working now

JBrummans, 2021-08-22:
Not sure if your still having this issue.
I experienced this as well. It may be related to your python PATH being incorrect. I simply reinstalled python and selected the option to add to PATH when installing. After that it worked.