cpwood/Pico-Go #76: Unable to connect to second attached Pico appearing as /dev/tty.usbmodem4

epmoyer, 2021-04-03:
## What are the steps to reproduce this issue?
1. Connect two Pico devices to a Mac.
2. The second appears (for me) as `/dev/tty.usbmodem4`.  The first is `/dev/tty.usbmodem0000000000001`.
3. Configure global settings as follows:
```
    "auto_connect": false,
    "manual_com_device": "/dev/tty.usbmodem4",
```

## What happens?
Pico-go reports "Address not configured. Please go to the settings to configure a valid address or comport"

## What were you expecting to happen?
Connect to the second Pico on `/dev/tty.usbmodem4`

## Any logs, error output, etc?

As a test, `rshell` can connect, so I am confident the second Pico is really at that dev node (i.e. at `/dev/tty.usbmodem4`):
```
% rshell -p /dev/tty.usbmodem4
Using buffer-size of 32
Connecting to /dev/tty.usbmodem4 (buffer-size 32)...
Trying to connect to REPL  connected
Retrieving sysname ... rp2
Testing if ubinascii.unhexlify exists ... Y
Retrieving root directories ... /framebuffer.bin/ /i2c_responder.py/ /i2c_responder_test.py/ /project.pico-go/
Setting time ... Apr 03, 2021 10:19:06
Evaluating board_name ... pyboard
Retrieving time epoch ... Jan 01, 1970
Welcome to rshell. Use Control-D (or the exit command) to exit rshell.
%
```

## Any other comments?

## Support info

With a single device attached and auto_connect enabled (so, device on `/dev/tty.usbmodem0000000000001`)

```
Pico-Go:      1.3.1
VS Code:      1.55.0
Electron:     11.3.0
Modules:      85
Node:         12.18.3
Platform:     darwin
Architecture: x64
Board:        Raspberry Pi Pico with RP2040
Firmware:     v1.14 on 2021-02-02 (GNU 9.3.0 MinSizeRel)
```

cpwood, 2021-04-03:
Thanks - I'll see if there's anything obvious jumping out at me, but I'm not in a position to debug it myself as I don't have two Picos. PRs gratefully accepted, if you're in a position to help!

epmoyer, 2021-04-03:
Sure; if nothing jumps out at you then I will poke though the code and see if I can figure it out/fix it on my side.  Thanks.

epmoyer, 2021-04-03:
Looks like only "auto_connect" True is working for me, even `"manual_com_device": "/dev/tty.usbmodem0000000000001"` fails, which I hadn't noticed before.  Trying to debug with print statements.  Does this.logger.silly() go somewhere or do I need to turn it on (and then where does it get logged to?).

Best I can tell, `this.autoconnectAddress` is not getting set for some reason (for me anyway).

epmoyer, 2021-04-03:
Ok, I can fix the problem by making the following change around line 384 of `lib/pymakr.js` (which as near as I can tell is the post-processed version of `src/pymakr.js`, and is the code which actually executes).

```
//.   MODIFIED THIS IF
//    if (!address && this.settings.auto_connect) {
      if (!address ) {
        var r = await this._getAutoconnectAddress();
        this.board.setAddress(r);
        address = r;
      } else {
//        REMOVED THIS IF (NO LONGER NECESSARY)
//        if (address) {
             this.board.setAddress(address);
//        }
      }
```

Maybe that's not the "right" way to fix it (I am not sure), but `._getAutoconnectAddress()` knows to look for/use `this.settings.manual_com_device` and returns it as the address if it exists, but (in my case) `._getAutoconnectAddress()`  was never getting called.  If that really is the "correct" fix, then I don't know how `manual_com_device` was working for anyone else (even in the single Pico case).  In my case, `manual_com_device` was being ignored (regardless of its value or how many Picos were attached).

epmoyer, 2021-04-03:
The fix above works for the "Connect" but on the second Pico (the one on `/dev/tty.usbmodem4`) doing an "Upload Project" fails. Something in that code path is using the default dev node I suspect, since it works on the other pico (presumably because the default turns out to be correct).  Trying to chase that code path down now too...

epmoyer, 2021-04-03:
[image]
Ah! I think it is just timing out. I don't know why, but the second Pico takes much longer than the first to show up in `/dev/` (maybe 5 seconds or so), so possibly the "upload bug" is just that the reconnect handler times out before the device shows back up again after a reset? I am going to see if I can find where I can increase the timeout just to test. Perhaps there is also some way on Mac to "fix" whatever issue is making the second Pico take so long to show up; that might be another way for me to attack this (and would be nice to fix, if possible).

epmoyer, 2021-04-03:
[image]
I was able to work around the reset/reconnect issue by increasing the sleep delay in the `.reset()` handler (in `shell.js`) to 6 seconds.  Perhaps a good solution (on Mac/Linux) would be to allow a longer time but poll for the `dev/` node to come back?  That would make it responsive in the "normal" case but permissive in the "abnormal" case.

My testing shows that on macOS:
- When plugging in either Pico alone, it shows up in `dev/` in under a second (about 0.8)
- When plugging in either Pico, waiting for it to enumerate, and then plugging the other, the second will take ~5 seconds (from thee second device's insertion) to show up in `dev/`

I replicated that timing behavior on two different machines, one Intel one M1, both running MacOS 11.2.1

epmoyer, 2021-04-04:
After doing a bunch of dual pico development this evening with the fix above in place it turns out that there is still at least one issue to resolve.  Sometimes one of the Picos never comes back (i.e. never shows up in /dev) after a reset following a "Upload Project".  I will try to chase down the root cause; possibly it is not even Pico-Go's "fault".

cpwood, 2021-04-09:
Thanks for all your investigation here! I'll try and include the suggested changes over this weekend. Did you figure out the missing device?

epmoyer, 2021-04-10:
I haven't figured it out yet. I am collaborating on a dual-pico project with a friend who is developing on Windows. This weekend he will kick the tires on Windows and see if he has related enumeration issues.  I suspect he won't but we'll see.  If need be I'll fire up my Windows box to try to replicate any issues he is having.

I've done some digging in the kernel/system logs but I couldn't find any indication of why the second Pico takes so long to enumerate. The only logging I could find related to the delay was in dmesg, which wasn't illuminating:

[image]

It surprises me that the first pico enumerates as `/dev/tty.usbmodem0000000000001` and the second as either `/dev/tty.usbmodem4` or `/dev/tty.usbmodem5`.  I don't know if the difference in zero padding and the skip from 1 to 4 (or 5) comes from the MacOS Kernel or from something in MicroPython. It might be interesting to plug two picos into a Linux box just to see how they enumerate, and whether the same delay issue exists.  I will try it on a VM to start; since that's easy and I don't have a Linux box sitting around at the ready.