cpwood/Pico-Go #77: Keyboard arrows do not work on the Pico console

thingslu, 2021-04-06:
## What are the steps to reproduce this issue?
1. Click on the Pico Console
2. Use the Up, Down, Left, Right keys from the keyboard

## What happens?
HPKM is typed at the console 

## What were you expecting to happen?
Move the cursor, access previous commands

## Any logs, error output, etc?
*(If it’s long, please paste to https://gist.github.com and insert the link here)*

## Any other comments?

## Support info

Copy this from the `Pico-Go > Help > Get support info` option in the command palette:

```
Pico-Go:      1.3.1
VS Code:      1.55.0
Electron:     11.3.0
Modules:      85
Node:         12.18.3
Platform:     win32
Architecture: x64
Board:        Raspberry Pi Pico with RP2040
Firmware:     v1.14 on 2021-02-05 (GNU 9.3.0 MinSizeRel)
```

cpwood, 2021-04-27:
This isn't something I'm able to reproduce, sorry; it works fine for me. Is it still a problem for you? Is there a particular scenario where it stops working, maybe?

thingslu, 2021-04-27:
That is funny, I tried on different computers and it never worked for me.
On the same computer with the pymakr plugin instead of Pico Go the arrow keys work fine. I only tried so far on Windows 10, I will test on Linux and let you know.