cpwood/Pico-Go #78: vscode for Pico with c

kevindawson, 2021-04-07:
## What are the steps to reproduce this issue?
1. set up the Pico toolchain

## What happens?
Pico-Go acts as if it is required

## What were you expecting to happen?
Pico-Go to act dead - **c** project

## Any other comments?
how do I disable Pico-Go for Pico **c** coding

Q how do I disable Pico-Go for none Pico - Micro Python projects

OR two pico projects one in &#181;py other in c on same machine

thanks in adv

_ps. vsc 1.55.0 has improved #51 #52 as prior screen data logging was an issue still in vsc but not Thorny_

cpwood, 2021-04-09:
Hi! Are you on version 1.3.1 of the extension? If so, I'd recommend setting the *global* config file so that `open_on_start` is `false`. Then, in your *project*'s `picogo.json`, have `open_on_start` is set to `true`. This should give you the behaviour you're looking for.

kevindawson, 2021-04-10:
@cpwood 
Thanks for the kick
picogo.json in PICO-EXAMPLES
```json
{
    "open_on_start": false,
    "statusbar_buttons": []
}
```
needed the [] to remove clutter from status bar

_Remembering the old adage "anything growing where you don't want it is a WEED"_ 

nb.
```text
Pico-Go:      1.3.1
VS Code:      1.55.0
Electron:     11.3.0
Modules:      85
Node:         12.18.3
Platform:     linux
Architecture: arm64
Board:        Raspberry Pi Pico with RP2040
Firmware:     v1.14 on 2021-04-09 (GNU 10.2.0 MinSizeRel)
```
So Long and thanks for all the fish!

closed 2021-04-27 by cpwood