cpwood/Pico-Go #79: Command 'picogo.initialise' not found after installing extension

patridge, 2021-04-09:
While VS Code shows the extension installed just fine, it appears as if the extension on Raspberry Pi OS isn't installing enough to work correctly.

## What are the steps to reproduce this issue?

1. [previously] Install latest MicroPython UF2 on new Pi Pico board
2. [previously] Update clean Raspberry Pi OS on Pi 4 board to latest
3. Install VS Code
4. Install Python 3
5. Install Pico-Go extension in VS Code
6. Load up [Pico-Go quick start](http://pico-go.net/docs/start/quick/)
7. Open empty folder for new project
8. Hit Ctrl+Shift+Alt+I

## What happens?

VS Code error pop-up: `command 'picogo.initialise' not found`.
Also, none of the Pico-Go items show in the bottom VS Code status bar.

## What were you expecting to happen?

I was expecting it to configure my project as the quick start describes.

## Support info

Copy this from the `Pico-Go > Help > Get support info` option in the command palette:

This also results in an error:

```
`Command 'Pico-Go > Help > Get support info' resulted in an error (command 'picogo.extra.getFullVersion' not found)
```

patridge, 2021-04-09:
For additional context, I get the same `command 'picogo.initialise' not found in VS Code on Windows 10.

cpwood, 2021-04-09:
Thanks! Can you confirm that you're on a 64-bit version of each OS, that you've got Python installed and within your PATH and that you're on version 1.3.1 of the extension?

Also, do you see anything in the Developer Tools console?

patridge, 2021-04-09:
Windows 10:
* Definitely 64-bit
* Python installed via Microsoft Store, doesn't appear to be in PATH, but I get a prompt when I type `python` in Cmd or PowerShell. I added the ugly Program Files\WindowsApps path to the folder containing the Python executable to my Path environment variable, just in case it helps.

Raspberry Pi OS on Pi 4:
* Possibly not 64-bit and not sure how to tell (`uname -m` outputs `armv7l`)
* Python possibly in PATH (`which python` returns `/usr/bin/python`)

I don't know what the Developer Tools console would be in VS Code? Is that some sort of terminal like the Debug Console?

When I relaunched VS Code just now, I did get an error pop-up, but I can't tell where it's coming from. It's possible that it's from the Pico-Go extension, but it could be that I broke PowerShell with a recent update too.

> The terminal process failed to launch: A native exception occurred during launch (File not found: ).

kevindawson, 2021-04-10:
@cpwood 
had you considered adding python version to - Get support info
see #78 for a Pi 4 support info
also Python 3 - virtual - pipenv

kevindawson, 2021-04-11:
@patridge

Hay man a 32bit os is not _cool_

If you had  installed a 64bit os - you would see the following:
```bash
$ uname -m
aarch64
```
also
```bash
$ cat /etc/lsb-release 
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=20.10
DISTRIB_CODENAME=groovy
DISTRIB_DESCRIPTION="Ubuntu 20.10"
```
if you had installed a 64bit os life is _groovy_ - even the Pico **c** toolchain works ;)

Oosl, 2021-04-15:
> Thanks! Can you confirm that you're on a 64-bit version of each OS, that you've got Python installed and within your PATH and that you're on version 1.3.1 of the extension?
> 
> Also, do you see anything in the Developer Tools console? 

I have same issue like this https://github.com/cpwood/Pico-Go/issues/79#issuecomment-816980639 on win10 64bit, pico-go 1.3.1, and install python3 by microsoft store.
after installed pico-go, I also got an error like this when launched vscode:
``` 
The terminal process failed to launch: A native exception occurred during launch (File not found: ).   
```
when I wanted to do anything by pico-go, I got following similar error messages:  
```
command 'picogo.toggleConnect' not found
command 'picogo.download' not found  
command 'picogo.reset.soft' not found
etc.
```

patridge, 2021-04-15:
And on the Raspberry Pi OS side of things, I just found out that 32-bit is the norm. The 64-bit version of the OS is still in its early days, despite the hardware support. So, maybe that's just not a happy path to go down, even with VS Code getting more attention on Raspberry Pi OS. Fortunately, the built-in Python IDE worked just fine.

kevindawson, 2021-04-15:
@patridge 

Raspberry Pi OS
64-bit version of the OS works ok
see above

life is groovy

cpwood, 2021-04-15:
It should be alerting you at startup that you have an incompatible OS. Happens within Activator.js from memory. Curious why that's not alerting you to the problem..

deenski, 2021-04-15:
@cpwood I'm having a hard time understanding how someone in this thread isn't violating your contribution guidelines with their low-quality, low-effort, obvious trolling? I don't want to violate the guidelines myself by naming or shaming. Feel free to reach out directly. My email is in my profile. I would have preferred to do so rather than leave this comment, but I don't have/want/need a twitter account. Sorry for the additional noise everyone.

cpwood, 2021-04-15:
Thanks @deenski .

Firstly, apologies for not being more responsive myself at the moment. Real life is getting in the way of any discretionary / hobby coding at the moment.

To make sure it's explicitly clear: **Pico-Go is only supported on 64-bit _OSs_ at the moment**. The Raspberry Pi situation complicates matters, since they're 64-bit devices that ship with 32-bit OSs unless you deliberately seek and install a 64-bit OS. I know that isn't always an immediately obvious thing for those who might be new to the Raspberry Pi world.

In terms of the substance of the problems in this issue.. I _suspect_ we've got two things going on:

1. a check on the extension's startup is not doing its job and alerting 32-bit OS users that Pico-Go is incompatible.  **If somebody can confirm that they're definitely _not_ seeing a message alerting you to incompatibility when you know you've got 32-bit OS, I'd appreciate a confirmation of the bug.**
2. continued issues identifying Python3, partly on Windows machines where Python is installed from the Microsoft Store, and partly where people are using environment-based Python setups.

Right now, Pico-Go is only compatible with vanilla Python out-of-the-box, though you can [manually set a Python path via the global settings](http://pico-go.net/docs/help/settings/#python_path). 

With regards to Python issues relating to the Microsoft Store (which should be supported and doesn't seem to be working), I'd appreciate any help on that while I'm a bit preoccupied with family matters. PRs are always gratefully received!

With regard to @deenski's point above if we're thinking about the same comments, I've read them differently . There _is_ some  useful feedback contained within them (pointing out how to spot that you've got 32-bit or 64-bit OS). I think there are some attempted puns in the responses (based on some of the output examples) that aren't translating culturally / immediately obvious / working at all though!

Oosl, 2021-04-16:
From my testing, I think the cause of this problem is the py launcher.  

Because I found that when I uninstalled all python and installed python through [the installer downloaded from the official website](https://www.python.org/ftp/python/3.9.4/python-3.9.4-amd64.exe) using the default options, pico-go worked fine.
[image]

When I tried to use the custom install option with the official installer, I found that the minimal installation option that would make pico-go work was to **just check py launcher and not even add python to the PATH.**  
[image]

The python installed via Microsoft store or scoop no doubt does **not have py launcher**.  
The py launcher installed via [the official installer](https://www.python.org/ftp/python/3.9.4/python-3.9.4-amd64.exe) does not recognize the python installed via Microsoft store or scoop.  

btw, the path of python via Microsoft store is  
```
C:\Users\[user_name]\AppData\Local\Microsoft\WindowsApps\[python_app_name]
```
the path of python via Scoop is
```
[scoop_path]\apps\python\current
```

deenski, 2021-04-16:
Can confirm the above work-around was successful for me. Uninstalled the store version, downloaded the installer from the website as suggested. Works like a charm. 

I have some free time and would love to help contribute to making the store version work, but have no idea where to start with that. If anyone here has some reading material that could nudge me in the right direction, I'll absolutely dedicate some time to it.

Thank you both to @cpwood and @Oosl and @kevindawson for your help so far.

stephanebertrand, 2021-04-23:
Hi,

on Windows 10 - latest stable MicroPython UF2 - Python 3.7.5 x64 (direct download, not store version )
i've similare errors : 'picogo.initialise' not found

Connection and execution are fine with Thonny, on USB COM3

cpwood, 2021-04-27:
With the Microsoft Store version of Python, it looks like that VS Code routine for launching a terminal does not have `python3` in its path, for some reason, even though it is in the path when you try it from a Command Prompt.

I've just tried hard-coding the full path to `python3.exe` and it worked absolutely fine with the Store version then..

Will see if I can figure it out!

cpwood, 2021-04-27:
Aaah.. it's happy with `python3.exe` but not with `python3`. I'll get .exe appended and see how we go!

closed 2021-04-27 by cpwood

cpwood, 2021-04-27:
`python3.exe` has been added explicitly to the list of executables to attempt (in addition to `python3`, which appears afterwards). Will be release 1.4.

utamadonny, 2021-05-10:
I have the same issues on Ubuntu 20.04 x86_64

kevindawson, 2021-05-10:
@utamadonny it's a libc version issue - if you had looked in `Log Extension Host` you would have seen the error.
ps. it also happens with Ubuntu 20.10 Architecture: x64 and vsc snap

EricPobot, 2021-05-22:
I have the same problem on Mint 19.3. No problem however for connecting to the Pico REPL from the terminal using the tty port.

@kevindawson where is the "Log Extension Host" you mention? I've searched in all menus but am an occasional VSCode user (more used to Vim for RPi stuff or PyCharm IDE for more serious work).

VSCode:
- Version: 1.56.2
- Commit: 054a9295330880ed74ceaedda236253b4f39a335
- Date: 2021-05-12T16:45:26.313Z
- Electron: 12.0.4
- Chrome: 89.0.4389.114
- Node.js: 14.16.0
- V8: 8.9.255.24-electron.0
- OS: Linux x64 4.15.0-143-generic

Pico-Go:
- Version: 1.4.2

System:    
- Host: Mint19 Kernel: 4.15.0-143-generic x86_64 bits: 64 compiler: gcc v: 7.5.0 
- Desktop: Cinnamon 4.4.8 wm: muffin dm: LightDM Distro: Linux Mint 19.3 Tricia  base: Ubuntu 18.04 bionic

kevindawson, 2021-05-22:
@EricPobot 
bottom window ( same as terminal)
top rhs - OUTPUT
then 
top lhs - drop down list - `Log (Extension Host)`

EricPobot, 2021-05-23:
@kevindawson 
Many thanks for guiding me. 

In the meantime I could access to the console log by using "Help/Toggle dev. tools" (Chrome dev tools obviously). Are they the same ?

According to the error messages found in both, the problem comes from the fact that the extension requires glibc 2.28+ and my current configuration has 2.27 only. It seems to be a bit strong as a requirement knowing that Mint 19 and distros based on the same Debian/Ubuntu base are not deprecated yet and are still widely used. It's installed on my work laptop and I don't plan to upgrade it yet since it's an LTS supported until April 2023.

BTW it seems to be common in VSCode extensions, since the CircuitPython one has the same problems.

However are these extensions really useful ? When using CircuitPython for instance, the actions they provide are easily doable by simply copying sources on the pseudo-disk or listing its content. This can even be added as custom tasks if really needed.

kevindawson, 2021-05-23:
@EricPobot my issue was caused by OS upgrade & SNAP: check your distro - eg. apt...

have fun

EricPobot, 2021-05-23:
@kevindawson 
My distro is updated regularly and works fine in all the other contexts I use the machines for.

At the end of the day, I'm more and more wondering if I'll not switch to my good old friends Vim and make, or PyCharm that I use for work ;) 

I confess having some difficulties with VSCode. It seems to be quite powerful, but also a bit too sensitive to external factors :/ I could observe this in the rest of JS world too: things can go havoc because when installing or updating some lib or tool, something has been touched although it should not have to.

Anyways, thanks for you support.

CircleOnCircles, 2021-07-27:
In 2021, WSL2 shows the same message. but just use it on Windows is fine.

heggland, 2022-04-14:
Getting the same error. using latest vscode and Python 3.10.4.
which python version is recommended?

marcellmueller, 2022-04-17:
Same error on Ubuntu 21.10 and Python 3.9.4 and VSCode .deb version installed.

The Pico works fine using Thonny so I'm not sure why it isn't connecting. I think I am done trying to debug this for now, I just want to mess with my new Pico and Thonny works but I use VSCode daily for work and enjoy all the features and plugins.

I was planning on doing a fresh install when Ubuntu 22.04 LTS releases on the 21 so I will try it again on the fresh install.

kevindawson, 2022-04-17:
@marcellmueller 
is this the same issue as #118 
if so try downgrading of VScode and Node.js
and update here please
thanks in adv.

EricPobot, 2022-04-17:
Downgrading (especially Node) has unfortunately side effects and breaks other things. Been there, seen it :(

After a first period being positively impressed by VSCode and envisioning making it my workhorse, I finally switched my mind after experiencing several problems similar to this one. There are often caused by some volatility and lack of stability of the JS and Node ecosystems, where things tend to randomly break now and then because some deeply buried dependency you were not aware of has been touched. 

I've quite often been faced to the best but also to the worst while working on JS and Node projects, due to the proficiency of JS devs and the fact that it seems to be a tendency for a lot of people to publish their work without enough review before (or even questioning about the real added-value, but this is another story). I've never encountered the same level of hazard WRT quality in Python for instance. I'm not saying everything is perfect there, far from it unfortunately, but I never experienced the same trouble, and I've far mode Python projects than JS/Node ones behind me.

Vi and Make have finally been (re)elected as my best allies for programming micro-controllers. Stable as rock, and paying back largely the efforts invested in their learning curve. OK, I confess having discovered them long before the VSCode adventure, so not really being a newbie there. This helps ;)

marcellmueller, 2022-04-17:
Downgrading Node is easy with NVM but downgrading VSCode isn't really a solution I am okay with so I will wait until Pico-Go is updated to work properly with it again.

@EricPobot thanks for your two cents but I'm not sure how most of this adds to the discussion.

EricPobot, 2022-04-17:
@marcellmueller you're right that it doesn't help solving the current issue, and I'm perfectly aware of this. I was thinking that providing one's feedback about alternate options aiming at being more productive (or at least spending less time on problems not belonging to the main track) may help. Sorry if I'm wrong, I promise not to talk about this anymore.

commit 45501ac0 (2021-04-27): Fix #79