cpwood/Pico-Go #80: Show Pico Pin Map - editor layout

kevindawson, 2021-04-15:
# Feature Request 🚀

Regardless of the editor layout - Show Pico Pin Map - always opens in default (top/left)

were as vsc opens a file next to the active editor.

could Pico-Go do the same with - Show Pico Pin Map

Thanks in Adv.

cpwood, 2021-04-27:
If I understand correctly, you're saying that the Pin Map always appears as the leftmost tab rather appearing to the right of the active tab.

If that's the case, that's not a behaviour I can reproduce; it appears to the right of the active tab, for me.

kevindawson, 2021-04-27:
@cpwood 
yes it's to the right - but in multi edit mode it always goes in -> top/left
where as vsc goes right of current editor

cpwood, 2021-04-27:
Aaah, understand, thanks. Will be fixed in 1.4.

commit 45e18e5e (2021-04-27): Fix #80
closed 2021-04-27 by cpwood