cpwood/Pico-Go #81: Raspberry Pi Support

CAProjects, 2021-04-22:
Please can you make some form of support for 32bit? so can be used on Raspberry Pi. I really don't like Thonny and would love to use VSCode instead on Raspberry Pi when programming pico devices

Kaydron1000, 2021-05-23:
I am having similar problems with raspberry pi. I did a fresh install of Raspberry Pi OS and still having issues. The issues i am having i think might be related to that "python" is set to 2.7 and "python3" is set to 3.7, but not sure.

Any command i try to use from Pico-Go or Pymakr does not execute. I get a response similar to this for all commands:
Command 'Pico-Go > Configure project' resulted in an error (command 'picogo.initialise' not found)

I've been hashing at this for a few hours now. I'm just trying to get intellisense and completion to work for my poor typing.

Kaydron1000, 2021-05-23:
After browsing some of the closed issues i found #72. This seems to be a similar issue. python3 --Version works but not py -V. Raspberry Pi OS initializes the system without the python launcher (although i didnt figure out yet if linux OS's even have a python launcher). 

I'll keep looking towards the python launcher though

CAProjects, 2021-05-23:
> After browsing some of the closed issues i found #72. This seems to be a similar issue. python3 --Version works but not py -V. Raspberry Pi OS initializes the system without the python launcher [...]

That issue appears to be related to windows. I have Python 3.7 as my default Python and py command and it does not work. I did read that the extension is 64 bit only which sucks for raspberry pi as there is no major 64 bit support on raspberry pi OS.

Not many developers are supporting it till it's released officially before the take it seriously

Kaydron1000, 2021-05-23:
I missed the documentation stating it was 64 bit. i'll put together the beta version and see if it works. it sounds like it will.

Kaydron1000, 2021-05-23:
Home page of http://pico-go.net/ states "PC (x64), Mac (x64 or Silicon) or Linux (x64 or arm64) computer required." so i see it now. Things are at least working now.

It would be a little helpful to add the following to dependencies in the README.md:
PC (x64), Mac (x64 or Silicon) or Linux (x64 or arm64) computer required.

CAProjects, 2021-05-23:
I use a pi-top[4] and their SDK for their pat form is 32bit and they are waiting for an official 64bit release of raspberry pi OS that is not beta and on top of that I wish the community would push them to focus development on the 64bit OS too so they can depreciate the 32bit.

The beta currently has little performance gain on the boards less than 8GB memory but there is a nori I left performance gain on 8GB boards

kevindawson, 2022-03-13:
Take a look at #56 
Please update here when you have tried with [arm64](https://www.raspberrypi.com/news/raspberry-pi-os-64-bit/) 
Thanks in adv.

kevindawson, 2022-04-02:
> Please update here when you have tried with [arm64](https://www.raspberrypi.com/news/raspberry-pi-os-64-bit/)

if you had LOOKED at Bullseye aka arm64 as I did you would have noticed it is working
@cpwood please close this issue