cpwood/Pico-Go #82: Disable Most Aggressive Keyboard Shortcut Thiefs

theAkito, 2021-04-23:
## What are the steps to reproduce this issue?
1. Install this extension.
2. Try to use a normally harmless shortcut like e.g. `CTRL + SHIFT + F`.
3. Notice, how this shortcut is taken over by this extension.

## What happens?
When this installation is installed, it takes over all default shortcuts for VS Code, as if Pico-Go were the only thing ever to be used in this VS Code instance.

## What were you expecting to happen?
This extension to have its own shortcuts or just keep them empty, so the user can decide how to bind the keys.

## Any logs, error output, etc?

[image]

## Any other comments?
I hope this was not intentional.

I have about ~120 Extensions installed in my VS Code and none of them ever took away crucial & basic keyboard shortcuts like Pico-Go did. This is a ridiculous but nevertheless a huge issue.

## Support info

Copy this from the `Pico-Go > Help > Get support info` option in the command palette:

I had to uninstall the extension, because I did not have the time to deal with this mess. I used the newest version, according to the VS Code Extension Marketplace.

cpwood, 2021-04-23:
99% of these have been inherited from the third-party pre-cursor project that came before Pico-Go: Pymakr. So no, not intentional on my behalf. Sorry you feel it's ridiculous!

Would anybody object to them being removed altogether?

kevindawson, 2021-04-26:
@cpwood 
had you considered it being an option that users could disable if they so wished in `pico-go.json`
there will always be conflict between plugins (keymapping)

theAkito, 2021-04-26:
> 99% of these have been inherited from the third-party pre-cursor project that came before Pico-Go: Pymakr. So no, not intentional on my behalf. Sorry you feel it's ridiculous!

Thanks for the information, good to know.

The thing is, as I already mentioned, I have over a hundred of other extensions and none of them took over such absolutely standard shortcuts. It's just weird and 100% unintuitive. Especially something like `CTRL + SHIFT + F` is something so standard, it's pretty much used in any IDE I have ever touched for searching across files.

> Would anybody object to them being removed altogether?

Therefore, I would suggest, that in this specific case it should get an urgent patch, where they are removed altogether, so we get this thing working quick. Once there will be a consensus of which default shortcuts make sense, they could be patched in afterward. However, in its current state, the VS Code of the user is immediately broken after installing this extension. That's why we need a quick fix, like removing it altogether, for now.

> there will always be conflict between plugins (keymapping)

It's not a harmless conflict between different extensions. It's a conflict between absolute standard shortcuts like `CTRL + SHIFT + F` and this specific extension. Of over 100 extensions I use, none of them are this intrusive.
Additionally, I have never seen/noticed a new extension take away shortcuts from a previously installed extension.

So, after they have been removed altogether, for now, it may be considered to use some non-default shortcuts or perhaps a shortcut chord, like e.g. `CTRL + SHIFT + ' & CTRL + SHIFT + F`. Then it wouldn't break the VS Code and the functionalities this extension provides would still remain intact, as in accessible through a shortcut.

cpwood, 2021-04-27:
Keybindings will be removed in the next release (1.4).

commit 82f6e5e4 (2021-04-27): Fix #82
closed 2021-04-27 by cpwood