cpwood/Pico-Go #83: buttons froze after reinstall cant initialise extention

MNS26, 2021-04-25:
## What are the steps to reproduce this issue?

i dont know what happened 
all the buttons became unresponsive 
reinstalling does not work

## What happens?
after install and running Pico-Go > Configure project i receive
`Command 'Pico-Go > Configure project' resulted in an error (command 'picogo.initialise' not found)`

## What were you expecting to happen?
the buttons to show up to interact with the Pi Pico

## Any logs, error output, etc?
*(If it’s long, please paste to https://gist.github.com and insert the link here)*

## Any other comments?
it gets stuck on installing 

## Support info

Copy this from the `Pico-Go > Help > Get support info` option in the command palette:

```
unable to get support info 
error message
`Command 'Pico-Go > Help > Get support info' resulted in an error (command 'picogo.extra.getFullVersion' not found)`
```
[image]

stuck on installing
[image]

after vscode restart and tried to configure project
[image]

cpwood, 2021-04-27:
Thanks for this - can you see if there is an error in VS Code's Developer Tools Console window?

Help > Toggle Developer Tools > Console tab

MNS26, 2021-04-27:
@cpwood there are a few warning and error 
[image]

MNS26, 2021-04-28:
@cpwood is there anything else i need to find (logs, files etc)?

pheidric, 2021-05-31:
Hi @cpwood ,
i've got the same problem after installing the plugin in my VS Code under Kubuntu 18.04.
There are no new buttons for the plugin, after **F1** > **Pico-Go > Configure Project** i got the error message:
`Command 'Pico-Go > Configure project' resulted in an error (command 'picogo.initialise' not found)`
Do you need any other information, some log foles or any thing else?