cpwood/Pico-Go #90: issue accessing a library from main.py

smith9, 2021-05-05:
## What are the steps to reproduce this issue?
1.have a main.py and a module file such as qwiic_vl53l1x.py
2.upload project with main.py having from qwiic_vl53l1x import QwiicVL53L1X as VL53L1X
3.run main.py

## What happens?
get error stating no module named 'qwiic_vl53l1x' - see below.

## What were you expecting to happen?
module class to be found and not report error.

## Any logs, error output, etc?
*(If it’s long, please paste to https://gist.github.com and insert the link here)*
>>> Running c:\Users\smith\Google Drive\dev\repos\micro_controllers\micropython\raspberry-pi-pico\tof-vl53l1x-qwiic\src\main.py
 
♦Traceback (most recent call last):
  File "<stdin>", line 4, in <module>
ImportError: no module named 'qwiic_vl53l1x'

## Any other comments?
I am new to micropython but am a developer by trade, so may well just be user error
I tried commenting out the offending import and adding log lines such as 
print("starting")
print(os.listdir("."))
which outputs (no mention of other module file alongside main.py)
>>> Running c:\Users\smith\Google Drive\dev\repos\micro_controllers\micropython\raspberry-pi-pico\tof-vl53l1x-qwiic\src\main.py

starting
['main.py', 'project.pico-go']

## Support info

Copy this from the `Pico-Go > Help > Get support info` option in the command palette:

```
Pico-Go:      1.4.2
VS Code:      1.55.2
Electron:     11.3.0
Modules:      85
Node:         12.18.3
Platform:     win32
Architecture: x64
Board:        Raspberry Pi Pico with RP2040
Firmware:     v1.14 on 2021-02-05 (GNU 9.3.0 MinSizeRel)
```

kevindawson, 2021-05-05:
> starting
['main.py', 'project.pico-go']

test.py
```python
import os
from qwiic_vl53l1x import QwiicVL53L1X as VL53L1X  # commenting out no issue,

print("starting")

# does not print out the filename for qwiic_vl53l1x.py
print(os.listdir("."))
```
qwiic_vl53l1x.py
```python
class QwiicVL53L1X:
    pass
```
in vcs
if I use "upload current file only" -> qwiic_vl53l1x.py
then "run" test.py on Pico
```
>>> Running test.py
 
starting
[..., 'project.pico-go', 'qwiic_vl53l1x.py']
```
have you read -> https://github.com/cpwood/Pico-Go/blob/main/CHANGELOG.md
are you up-to date?
ps. next time it would help if you supplied the test code :((

smith9, 2021-05-06:
Hi,
TL;DR - seems to be a py file length issue if a file is 63k

version of pico-go was latest 1.4.2, the version of VSCode was 1.55.2, I updated to 1.56.0.
I tried the shorter version of files and eventually got them to work

I did go through a spate of log lines like
```
[1/2] Writing file 'main.py' (1 kB)
[2/2] Writing file 'qwiic_vl53l1x.py' (63 kB)
Upload failed.Hashes do not match between computer and board.
undefined
undefined
```
and 
```
[1/2] Writing file 'main.py' (1 kB)
[2/2] Writing file 'qwiic_vl53l1x.py' (63 kB)
Upload failed.Not enough memory available on the board.
Not enough memory available on the board.
Not enough memory available on the board.
```
I tried stripping out comments from the python lib and got the file down to 33kB
```
[1/1] Writing file 'qwiic_vl53l1x.py' (33 kB)
Upload done, resetting board...
OK
```
I can't remember getting any errors yesterday relating to file sizes (not saying they weren't there)

So looks like there may be a limit of file size somewhere between Pico-Go and the Pico

The library now sort of works (it allows you to pass in a I2C instance, but the one I am passing (machine.I2C) does not have quite the same interface as the sparkfun qwuiic interface which is a shame they all couldn't use the same interface.

I am trying to find a python library for the vl53l1x time of flight sensor. So more digging to do...

Thanks

kevindawson, 2021-05-06:
please try the ftp option -> http://pico-go.net/docs/start/quick/#managing-files-via-ftp
would like to know if Writing file 'qwiic_vl53l1x.py' (63 kB) works or larger
thanks

smith9, 2021-05-06:
Looks like perhaps the Pico does not like files larger than about 63k
```
>>>
Started FTP server: ftp://pico@127.0.0.1:2121

Stopped FTP server.
```
uploaded 63kb file using filezilla
then tried to run main a few times
first time - import error
```
>>> Running c:\Users\smith\Google Drive\dev\repos\micro_controllers\micropython\raspberry-pi-pico\tof-vl53l1x-qwiic\src\main.py

♦Traceback (most recent call last):
  File "<stdin>", line 4, in <module>
ImportError: no module named 'qwiic_vl53l1x'
```
Second time same error
```
>>> Running c:\Users\smith\Google Drive\dev\repos\micro_controllers\micropython\raspberry-pi-pico\tof-vl53l1x-qwiic\src\main.py

♦Traceback (most recent call last):
  File "<stdin>", line 4, in <module>
ImportError: no module named 'qwiic_vl53l1x'
```
so looks like 63k file not ok
I then get the file again from https://raw.githubusercontent.com/sparkfun/Qwiic_VL53L1X_Py/master/qwiic_vl53l1x.py
and paste to vscode, it is now 55kb rather than 63kb (has tabs from sparkfun, and at some point on my pc it is getting converted to spaces and increasing in size)
So pico-go can't upload 63k files
Filezilla and pico-go together can
but the file seems unusable - reported as module not found.

```
>>>
Uploading current file (qwiic_vl53l1x.py)...
Not safe booting, disabled in settings

Uploading to / ...
Reading file status
Uploading single file
[1/1] Writing file 'qwiic_vl53l1x.py' (56 kB)
Upload done, resetting board...
OKPlease connect your device
Please connect your device

Connection attempt cancelled
Connecting to COM3...

>>> Running c:\Users\smith\Google Drive\dev\repos\micro_controllers\micropython\raspberry-pi-pico\tof-vl53l1x-qwiic\src\main.py
```
below seems weird, still reporting as not able to import module
```
♦Traceback (most recent call last):
  File "<stdin>", line 4, in <module>
ImportError: can't import name QwiicVL53L1X
♦>
>>> Running c:\Users\smith\Google Drive\dev\repos\micro_controllers\micropython\raspberry-pi-pico\tof-vl53l1x-qwiic\src\main.py
```
try again below and not importing again
```
♦Traceback (most recent call last):
  File "<stdin>", line 4, in <module>
ImportError: can't import name QwiicVL53L1X

>>>
raw REPL; CTRL-B to exit
>
Running c:\Users\smith\Google Drive\dev\repos\micro_controllers\micropython\raspberry-pi-pico\tof-vl53l1x-qwiic\src\main.py
```
Haven't changed anything and on 3rd attempt (no change to source) it finds the module and complains about interface difference between machine.I2C that I am passing from main.py to the library (getting the I2C interface error is expected I guess, trying to find one that is a python one and uses ideally machine.I2C, other one I found needs smbus2)

not sure yet why it ran twice with one error then a third time it worked, I didn't knowingly change anything
```
MPY: soft reboot
KeyboardInterrupt:
♦Traceback (most recent call last):
  File "<stdin>", line 4, in <module>
  File "qwiic_vl53l1x.py", line 114, in <module>
ImportError: no module named 'qwiic_i2c'

```
Main.py added .txt to get it to upload [main.py.txt](https://github.com/cpwood/Pico-Go/files/6434763/main.py.txt)
My lib file is https://raw.githubusercontent.com/sparkfun/Qwiic_VL53L1X_Py/master/qwiic_vl53l1x.py

kevindawson, 2021-05-06:
This is what I get: upload current file only
```text
>>> 
Uploading current file (qwiic_vl53l1x.py)...

Uploading to / ...
Not safe booting, disabled in settings
Reading file status
Uploading single file
[1/1] Writing file 'qwiic_vl53l1x.py' (55 kB)
Upload done, resetting board...
OK
```
then: Run
```text
>>> Running test.py
 
Traceback (most recent call last):
  File "<stdin>", line 3, in <module>
  File "qwiic_vl53l1x.py", line 114, in <module>
ImportError: no module named 'qwiic_i2c'
 
>>> 
```
Looks to me as thought vsc + pico-go + pico all ok ;)

ps. 1.15 -> https://micropython.org/download/rp2-pico/
@smith9 
- This is on you: ImportError: no module named 'qwiic_i2c'
  - import qwiic_i2c	# I2C bus driver package
    - `machine.I2C`
- But the file size v transfer I agree with.
  - Please change Title to reflect this

ps. take a looked at: `machine.soft_reset()`

@cpwood ping
if I make a big.py ( 2 x qwiic_vl53l1x.py)
pico-go
- upload current file only -> fails
- ftp -> fails

Thonny 3.3.7 works ok

looks like a Pico-Go file transfer > 64kiB issue

smith9, 2021-05-07:
@kevindawson `Looks to me as thought vsc + pico-go + pico all ok ;)`
yes I agree with that up to 63k- that file https://raw.githubusercontent.com/sparkfun/Qwiic_VL53L1X_Py/master/qwiic_vl53l1x.py
is only 56kb when tabs are used for indenting, but if an editor (Intellij for me at one point) can convert tabs to spaces which resulted in a 63kb file for the same logical content

`Thonny 3.3.7 works ok`
that is interesting, perhaps the pico can handle files larger than 63k - will need to try that.

`This is on you: ImportError: no module named 'qwiic_i2c'`
Yes, I know, was just putting that in for context. I had commented out the qwiic_i2c because I wanted to try machine.I2C

Thanks

smith9, 2021-05-07:
tried the 63k version uploaded by thonny and that worked.

kevindawson, 2021-05-07:
@smith9 
read this -> https://github.com/sparkfun/Qwiic_I2C_Py/pull/3
go fork n build, have fun