cpwood/Pico-Go #95: Pico-Go -> Configure Project creates empty .vscode folder

GregWoods, 2021-07-05:
Pico-Stubs folder is not created under .vscode, and so intellisense doesn't work. 
There is no indication in the docs or readme that I have to install pico-stubs separately.

Tried running VS Code as admin and standard user. 

Not sure where to go from here

## Support info

python 3.9.6

Pico-Go:      1.4.2
VS Code:      1.57.1
Electron:     12.0.7
Modules:      87
Node:         14.16.0
Platform:     win32
Architecture: x64
Board:        Raspberry Pi Pico with RP2040
Firmware:     v1.16 on 2021-07-02 (GNU 10.2.0 MinSizeRel)

GregWoods, 2021-07-05:
Posting an issue is a sure fire way to find the answer!
I had a lot of VSCode extensions (all disabled)
Uninstalling several of them, including PyMaker caused intellisense to start working.

It looks like the Pico-Stubs folder is a red-herring!

GregWoods, 2021-07-05:
Correction intellisense is broken for me, but I no longer get the yellow squiggles underlining my imports

GregWoods, 2021-07-05:
**SOLVED**: I had recently started using a network share for developing code. This was causing the issue.
Once I went back to a sub-folder of my "home" folder on C: , everything worked.
This may be considered a bug - it is certainly an annoying restriction. But the workaround is so easy, I suspect this will be closed as too low a priority.

Tweekazoid, 2021-07-17:
I also experience this bug,
running project from NAS (aka. \\someplace\pico_project)
maybe limitation of win32 and network drive linking (sym-links, hard-links)? 
----------------------

Pico-Go:      1.4.2
VS Code:      1.58.1
Electron:     12.0.13
Modules:      87
Node:         14.16.0
Platform:     win32
Architecture: x64
Board:        Raspberry Pi Pico with RP2040
Firmware:     v1.16 on 2021-06-18 (GNU 10.2.0 MinSizeRel)

----------------------
workaround is to "Configure project" on local drive, and than copy files manually

closed 2021-08-07 by cpwood