cpwood/Pico-Go #96: Extension Fails to Load on Ubuntu 20.04.2 + Visual Studio Code (Snap)

havenith, 2021-07-23:
## What are the steps to reproduce this issue?
1. Install pico-go from marketplace
2. Reload Visual Studio Code

## What happens?
Error in Log (Extension Host), pico-go fails to load properly. When invoking pico-go configure project, get error: `command 'picogo.initialise' not found`

## What were you expecting to happen?
Pico-go extension is available to use

## Any logs, error output, etc?
Log (Extension Host):
```

[error] Error: /snap/core/current/lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.28' not found (required by /home/simon/.vscode/extensions/chriswood.pico-go-1.4.2/node_modules/@serialport/bindings/lib/binding/node-v87-linux-x64/bindings.node)
	at process.func [as dlopen] (electron/js2c/asar_bundle.js:5:1846)
	at Object.Module._extensions..node (internal/modules/cjs/loader.js:1185:18)
	at Object.func [as .node] (electron/js2c/asar_bundle.js:5:1846)
	at Module.load (internal/modules/cjs/loader.js:982:32)
	at Module._load (internal/modules/cjs/loader.js:823:14)
	at Function.f._load (electron/js2c/asar_bundle.js:5:12913)
	at Function.i._load (/snap/code/70/usr/share/code/resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:103:27843)
```

## Any other comments?
The version of libc6 in snap core is 2.23, so the error on not finding 2.28 is understandable. But why would `serialport` need a later version? And why would the snap version of libc not be at the correct version for Visual Studio Code installed via snap? I do have a later version of libc on my machine (2.31) at `/lib/i386-linux-gnu/libc.so.6`, but that is not used by the serialport bindings it seems.

## Support info

``` 
/snap/core/current/lib/x86_64-linux-gnu/libc.so.6
GNU C Library (Ubuntu GLIBC 2.23-0ubuntu11.3) stable release version 2.23
```

Visual Studio Code details
```
Version: 1.58.2
Commit: c3f126316369cd610563c75b1b1725e0679adfb3
Date: 2021-07-14T22:30:16.440Z
Electron: 12.0.13
Chrome: 89.0.4389.128
Node.js: 14.16.0
V8: 8.9.255.25-electron.0
OS: Linux x64 5.4.0-80-generic snap
```

karye, 2021-07-26:
Same here:

> mainThreadExtensionService.ts:95 Activating extension 'ChrisWood.pico-go' failed: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.28' not found (required by /home/user/.vscode/extensions/chriswood.pico-go-1.4.2/node_modules/@serialport/bindings/lib/binding/node-v87-linux-x64/bindings.node)

My GLIBC is 2.27 on Ubuntu:

> ldd (Ubuntu GLIBC 2.27-3ubuntu1.4) 2.27
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
Written by Roland McGrath and Ulrich Drepper.

Why is 2.28 needed?
Is there a way to specify 2.27 or lower?

flipphillips, 2021-08-03:
Same here. 

`/snap/core/current/lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.28' not found (required by /home/flip/.vscode/extensions/chriswood.pico-go...`

This is with normal VS Code and not Insiders (like the macOS bug)

Same versions as @karye

cpwood, 2021-08-07:
Hi all - might be slightly outside of my comfort zone here, but serialport needs to be recompiled for Electron. To do that, we use [Electron-Rebuild(https://github.com/electron/electron-rebuild). 

We run this on the `ubuntu-latest` GitHub [virtual environment](https://github.com/actions/virtual-environments/blob/main/images/linux/Ubuntu2004-README.md) which is based on Focal.

So it's not so much that _serialport_ needs that version, it's that it was built on an agent that used it. The older Ubuntu agents are deprecated though, so I'm not sure where to go with this.

Input is appreciated!

natcl, 2021-09-12:
I have the same issue, any workarounds ?
Thanks !

cdrcqnts, 2021-11-05:
Same issue here running Ubuntu 20.04 LTS

cdrcqnts, 2021-11-05:
The issue on Ubuntu seems to be related to the installation of vs code via snap.
This workaround worked for me: 
1. Uninstall the vs code
2. Install vs code from the .deb package from the offial website

cdrcqnts, 2021-11-05:
> I have the same issue, any workarounds ? Thanks !
> bump - same with 21.04

The issue on Ubuntu seems to be related to the installation of vs code via snap.
This workaround worked for me:

    Uninstall the vs code
    Install vs code from the .deb package from the offial website

flipphillips, 2021-11-05:
Agree with @cdrcqnts - that seemed to have fixed it for me. Can't install via `snap` nor the Ubuntu App Store - which, I dunno, uses snap?

havenith, 2021-11-06:
I spent a bit of time on this trying to understand how electron-builder can be made to compile/link to an older GLIBC but ultimately gave up after running down a rabbit hole and hitting a problem in nodejs's latest headers, which affect gyp, which is used by electron-builder, which is needed to compile serialport to native binaries (https://github.com/nodejs/node-gyp/pull/2497). 

It still comes down to the fact that VSCode when installed via snap will try to use the old GLIBC in the snap structure to execute serialport. Why this snap version has to be so far behind I still haven't figured out. Any ideas? But ultimately to play nicely with snap, I think the proper solution would be to compile on an Ubuntu 16 to match snap's old GLIBC.

Anyway, if you don't install VSCode via snap, e.g. a deb package as per @cdrcqnts, the non-snap GLIBC will be used, and that should work if your distribution is somewhat recent. 
@flipphillips it's probably easier to just [download](https://code.visualstudio.com/download) the deb file and install via terminal:

`sudo dpkg -i /<pathtodownloads>/code_1.62.0-1635954068_amd64.deb`

havenith, 2021-11-09:
There are workarounds as described above (don't use VSCode snap version, or build serialport bindings on Ubuntu 16) but after some more digging I think I found the offending symbol linked to during compilation of SerialPort on newer Ubuntus e.g. 20+. 
After running an objdump on the resulting binaries for a Ubuntu16 and 21 versions, I found that the newer versions linked in fcntl64 which makes the binary require GLIBC_2.28, the Ubuntu 16 version uses fcntl which only requires  GLIBC_2.2.5.

Ubuntu16
`objdump -T./node_modules/@serialport/bindings/bin/linux-x64-89/bindings.node | grep fcntl`
gives:
`0000000000000000      DF *UND*  0000000000000000  GLIBC_2.2.5 fcntl`

Ubuntu21
`objdump -T ./node_modules/@serialport/bindings/bin/linux-x64-64/bindings.node | grep GLIBC_2.28`
gives
`0000000000000000      DF *UND*	0000000000000000  GLIBC_2.28  fcntl64`

Curious if one can't force a link to the older 'fcntl' I landed [here](https://stackoverflow.com/questions/58472958/how-to-force-linkage-to-older-libc-fcntl-instead-of-fcntl64) and [here](https://forum.rebol.info/t/dynamic-linking-static-linking-and-the-march-of-progress/1231). Interesting background for the curious.
Although it seems we're not alone in our disappointment with this surprising dependency on libc 2.28, there doesn't seem to be a reasonable serialport compilation change that doesn't lead to a maintenance nightmare. The fix is probably in upgrading the VSCode-snap-version shared libc.