cpwood/Pico-Go #97: Failure to load, vscode insiders, macOS 11.5.1

flipphillips, 2021-07-29:
## What are the steps to reproduce this issue?
1. Install latest plugin
2. Attempt to use any `pico-go` commands

## What happens?
Errors with command not found.

## What were you expecting to happen?
It to actually work :)

## Any logs, error output, etc?

From Extension Server Logs - 

[2021-07-29 16:36:12.037] [exthost] [error] Activating extension ChrisWood.pico-go failed due to an error:
[2021-07-29 16:36:12.037] [exthost] [error] Error: Could not locate the bindings file. Tried:
...
→ /Users/xxx/.vscode-insiders/extensions/chriswood.pico-go-1.4.2/node_modules/@serialport/bindings/lib/binding/node-v89-darwin-x64/bindings.node
	at bindings (/Users/fxpppr/.vscode-insiders/extensions/chriswood.pico-go-1.4.2/node_modules/bindings/bindings.js:126:9)
	at Object.<anonymous> (/Users/fxpppr/.vscode-insiders/extensions/chriswood.pico-go-1.4.2/node_modules/@serialport/bindings/lib/darwin.js:2:36)
	at Module.u._compile (/Applications/Visual Studio Code - Insiders.app/Contents/Resources/app/out/vs/loader.js:4:1315)
...
	at v._activateExtension (/Applications/Visual Studio Code - Insiders.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:90:11704)
	at processTicksAndRejections (internal/process/task_queues.js:93:5)
	at async Promise.all (index 2)
	at async Promise.all (index 0)

## Any other comments?

## Support info

Can't evaluate because won't load.
```
```

flipphillips, 2021-07-30:
So, a little digging seems to be a node/typescript version collision. My really naive hypothesis is that insiders is using a newer version of typescript that the node serialport doesn't support?

cpwood, 2021-08-07:
Pico-Go needed new bindings for modules v89. These will be included in the next release.

closed 2021-08-07 by cpwood