cpwood/Pico-Go #98: Pico-Go works only until the first restart of VSC, then full reinstall is required to work again - Windows 10

WojTechnical, 2021-08-06:
## What are the steps to reproduce this issue?
1. Install latest Visual Studio Code(1.59.0 here) on Windows 10(20H2 version here)
2. Install Pico-Go extension through marketplace
3. Observe that the extension works correctly, all functionality is there and functional
4. Close VSC, reopen again, notice that the bottom bar for Pico-Go is not there

## What happens?

Pico-Go doesn't load after first restart of VSC. Reinstalling Pico-Go doesn't fix it, the only fix I found is to remove VSC entirely from Windows, then install it again, then install the extension again - and it works....until the first restart.

## What were you expecting to happen?

Pico-Go to work all the time, not only until the first restart.

## Any logs, error output, etc?

Extension logs here:
https://gist.github.com/WojTechnical/d0d694521d99355b431d0615d1a86809

## Any other comments?

## Support info

Copy this from the `Pico-Go > Help > Get support info` option in the command palette:

Pico-Go:      1.4.2  
VS Code:      1.58.2 
Electron:     12.0.13
Modules:      87     
Node:         14.16.0
Platform:     win32  
Architecture: x64    
Board:        Raspberry Pi Pico with RP2040
Firmware:     v1.16 on 2021-06-18 (GNU 10.2.0 MinSizeRel)

WojTechnical, 2021-08-06:
To add to this - I replicated the issue on my laptop as well, so it's not limited to my main PC. Same error logs too.

WojTechnical, 2021-08-06:
So I've tried just giving it one of the bindings.node files, by copying contents of

C:\Users\gambi\.vscode\extensions\chriswood.pico-go-1.4.2\node_modules\@serialport\bindings\lib\binding\node-v87-win32-x64

into:

C:\Users\gambi\.vscode\extensions\chriswood.pico-go-1.4.2\node_modules\@serialport\bindings\build

But then I just get this error when starting VSC:

[2021-08-06 16:35:34.248] [exthost] [error] Error: The module '\\?\c:\Users\gambi\.vscode\extensions\chriswood.pico-go-1.4.2\node_modules\@serialport\bindings\build\bindings.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 87. This version of Node.js requires
NODE_MODULE_VERSION 89. Please try re-compiling or re-installing

So yeah, it looks like the extension is looking for node v89 extension, but visual studio installs v87 by default.

So.....why does this happen? And how can I update the node bindings manually??

WojTechnical, 2021-08-06:
Ok, I think I found out the reason why it worked at all:

I used the install file for VS Code 1.58.2, and Pico-Go extension comes with SerialPort v87 which is compatible with that version of VSC. Extension works correctly.

However, as soon as I close it the first time, it auto-updates to 1.59.0, which isn't compatible with this extension anymore, so any subsequent restart of VSC breaks the extension. Only by fully uninstalling VSC and reinstalling 1.58.2 I can get the extension to work again, until the very first restart when VSC upgrades itself and breaks the extension.

WojTechnical, 2021-08-06:
Ok, I can confirm that disabling VSC auto-update, and installing 1.58.2 stops the extension from breaking on every subsequent restart. Don't think this resolves the bug, but it's a workaround for now.

barbeque, 2021-08-06:
I'm having a similar problem (same stack trace on the serialport node module) on macOS 10.15.7 with VSCode 1.59.0.

cpwood, 2021-08-07:
Hi all - thanks for this, closing as a duplicate of #97.

closed 2021-08-07 by cpwood