cpwood/Pico-Go #99: Compatablity issue - VS Code 1.59.0...

kevindawson, 2021-08-06:
## What are the steps to reproduce this issue?
1. upgrade vsc 1.59.0

## What happens?
absolutely nothing

## What were you expecting to happen?
vsc upgrade and Pico-go in harmony

## Any other comments?
Keep up the good work, its just a mater of time

## Support info

Pico-Go:      1.4.2
VS Code:      1.58.2
Electron:     12.0.13
Modules:      87
Node:         14.16.0
Platform:     linux
Architecture: x64
Board:        Raspberry Pi Pico with RP2040
Firmware:     v1.16 on 2021-06-29 (GNU 10.2.0 MinSizeRel)

WojTechnical, 2021-08-06:
Yep, it's the same bug as the one I created earlier, 1.59.0 isn't compatible unless the bindings.node file gets updated to support it....

https://github.com/cpwood/Pico-Go/issues/98

cpwood, 2021-08-07:
Closing as a duplicate.

closed 2021-08-07 by cpwood

cpwood, 2021-08-07:
Fixed v.1.43.

kevindawson, 2021-08-07:
@WojTechnical sorry for the duplicate issue
@cpwood my bad