cpwood/Pico-Go #102: VSCode not connecting to pico, stuck in "connecting to COM3..."

SiiNerGia, 2021-09-02:
## What are the steps to reproduce this issue?
1.Open VSCode with pico connected, it will try to connect to it automatically

## What happens?
[image]

## What were you expecting to happen?
seeing the >>> and being able to run code properly. At the moment trying to run code does nothing, however, if I press the reset button, and then try to run some code, it works the first time, but as soon as I stop it and rerun it, it doesn't.

## Any logs, error output, etc?
Made sure my COM port was only being used by VSCode:
[image]

I have already rebooted and relaunched all apps with no success

## Any other comments?

## Support info

```
Pico-Go:      1.4.3
VS Code:      1.59.1
Electron:     13.1.7
Modules:      89
Node:         14.16.0
Platform:     win32
Architecture: x64
Board:        Raspberry Pi Pico with RP2040
Firmware:     v1.16 on 2021-06-18 (GNU 10.2.0 MinSizeRel)
```

smeyer87, 2021-09-12:
Unsure if the same issue but similar.  I get a similar 'Connecting to COM3' message and timeout when I know that my Pico is on COM5.  When I change the config settings per the documentation:

"auto_connect":false,"manual_com_device":"COM5"

then I get the message: 
"Address not configured. Please go to the settings to configure a valid address or comport"

I am sure that I'm using the correct config file because if I switch auto_connect: true then I get the COM3 error again.

Listing serial ports shows COM5 in addition to the others:
_Found 4 serialports
COM3 (Microsoft) (copied to clipboard)
COM5 (Microsoft)
COM4 (Microsoft)
COM1 ((Standard port types))_

x509cert, 2021-11-06:
FWIW, I have the same issue on Win11 - my device is on COM10, but pico-go tries connecting to COM5, if I configure it to connect to COM10, however, my issue is I did not have the auto_connect set to false; once I set it to false it worked fine

megazhuk, 2021-11-18:
I have this message too - "Address not configured. Please go to the settings to configure a valid address or comport"
Is anyone solve it?

garymeg, 2022-01-30:
I have same Problem with windows 10 but im getting a different error
I get "Cannot read property 'sleep' of undefined" and then in terminal i get "Connection failed"
Thonny works fine and sees the pico

Please find a solution as i like vscode and not thonny.

gvkrum, 2022-02-11:
в настройках поменял "open_on_start": true, "auto_connect": false, "manual_com_device": "COM5"
1. подключаем pico
2. запускаем VSCode
это решение проблем с большинством описанных в теме ошибок.

markjhoy, 2022-03-10:
> in the settings I changed "open_on_start": true, "auto_connect": false, "manual_com_device": "COM5" connect pico Run VSCode is a solution to the problems with most of the errors described in the topic.

Can confirm this worked for Windows 10. I originally had my pico on COM6, installed a new bluetooth module that took up COM6, and my Pico was showing on COM4. It wasn't until I tried the instructions here that I was able to get it to switch to COM4 (the key was the `"open_on_start": true` setting and restarting VS Code)