cpwood/Pico-Go #104: Cannot connect Pico-console when main.py is running.

Lak4CYUT, 2021-09-28:
## What are the steps to reproduce this issue?
1. Upload a main.py include an infinite loop.
2. Tune off power then tune on.
3. Click "Pico Disconnected" to connect Pico.

## What happens?
Pico-Go Console stuck on "Connecting to /dev/ttyACM0...".
And cannot send 'ctrl-c' to stop running program.

## What were you expecting to happen?
Send 'ctrl-c' to stop running program and connect to device.
(Maybe provide a focus-stop function)
**IDE thonny can stop running program and connect to device with 'stop/reset backend' button.**

## Any logs, error output, etc?
*(If it’s long, please paste to https://gist.github.com and insert the link here)*

## Any other comments?

## Support info
```
Pico-Go:      1.4.3
VS Code:      1.59.0
Electron:     13.1.7
Modules:      89
Node:         14.16.0
Platform:     linux
Architecture: x64
Board:        Raspberry Pi Pico with RP2040
Firmware:     v1.17 on 2021-09-02 (GNU 10.2.0 MinSizeRel)
```

varofla, 2021-10-15:
When you upload main.py file to pico, it automatically reads and starts "main.py" file when power is on.
This is because you cannot connect to it.
If you want to connect, you should erase flash by uploading "nuke.uf2" file.
You can download "nuke.uf2" file to the link below.
https://learn.adafruit.com/getting-started-with-raspberry-pi-pico-circuitpython/circuitpython

I'm not familiar with English, so I'm not sure if it's expressed correctly.
Please refer to the link below for more details.
https://forums.raspberrypi.com/viewtopic.php?t=300903

Lak4CYUT, 2021-10-16:
@varofla 
As my commented above, IDE thonny CAN stop it normally, that is why I think that is a issue.
I guess thonny doesn't check the connection status, just send the force stop command to device. 

(If we need to erase flash every time when we developing, I think it is a little bit ... you know ..)

cebess, 2022-03-07:
I am seeing the same issue. I also notice that Thonny disconnects my bluetooth keyboard when it starts up on the Raspberry Pi.

MichaelMichaelMichaelMichaelMichael, 2022-04-05:
With VS Code and Pico-Go you can upload this file
https://cdn-learn.adafruit.com/assets/assets/000/099/419/original/flash_nuke.uf2?1613329170
After that you can upload your uf2 as usual.

Lak4CYUT, 2022-04-12:
Yes, it is interesting. I don't know why Thonny works so smooth. 
But I really want to dev with VSCode....

v3xation, 2022-04-12:
Ah... [#7867](https://github.com/micropython/micropython/issues/7867) over on micropython since 1.17. Looks to be a bad interaction with CTRL-A followed by CTRL-C related to the VCP over USB. Tested [v1.18-330-gd242a9b7f (2022-04-12) .uf2](https://micropython.org/resources/firmware/rp2-pico-20220412-unstable-v1.18-330-gd242a9b7f.uf2) and it appears to be working more or less correctly, it will appear to hang on connect CTRL-C to get repl to pop up. The only thing that seems off is that if you had a print loop running you won't see the output when reconnecting.