cpwood/Pico-Go #108: Help Wanted - Maintenance of Pico-Go

cpwood, 2021-10-20:
Hi everyone,

As you've probably sensed from my lack of presence here of late, my ability to maintain Pico-Go as a project is very much reduced because of various things going on in my private life outside of GitHub! It's proving to be a tough 2021.

At the moment, I'm consciously doing the bare minimum - which amounts to simply ensuring that Pico-Go stays compatible with emerging versions of VS Code. Not only am I not spending time on Pico-Go but neither am I able to spend my time on enjoyable Pico projects either!

The best way you can help move things along at the moment is to offer pull requests for any fixes you're able to perform yourselves. Unfortunately, creating issues and waiting for me to resolve them personally is probably going to result in a long wait for the foreseeable future.

If anybody would like to get involved in administering the project - or even take it over - please do get in touch for a chat.

All the best,

Chris

SteveGoodenough, 2022-02-10:
Hope all better in 2022 Chris, but seeing the above I'll defer to posting my issue (related to GLIBC dependency).

Shame you've not had any offers, my js is basic to say the least but I'd love to have helped get this supported.

kevindawson, 2022-03-13:
@cpwood how's it hanging!
I think it's time to look at [arm64](https://www.raspberrypi.com/news/raspberry-pi-os-64-bit/)
I have asked #81 for feedback

@SteveGoodenough it's open source - just join in where you can.
I help with Linting & testing

tofylion, 2022-03-22:
@cpwood I've fixed a recurring issue in windows where the arrow keys seem to be crashing the extension. If you want to take a look and check if it's okay, I'll be really happy. Thanks for the great extension!

slowpokefarm, 2022-05-02:
It seems there will be no compatibility fix for the latest vscode without someone taking over this project. Does anyone know a working alternative?

chrisfu, 2022-05-17:
@pyre

> 

https://github.com/pryre/Pico-Go