cpwood/Pico-Go #113: No Pico-Go Commands Found

HopWorks, 2022-01-06:
## What are the steps to reproduce this issue?
1. Persistent and does not resolve when conditions change, as in... 
2. VSCODE restart
3. Windows restart
4. UF2 reinstall
5. UF2 reinstall after flash wipe
6. Another PI PICO device used
7. Different USB port used
8. Cable change
9. Extension uninstalled and reinstalled.

## What happens?
Session worked fine, was able to upload, stop, start, connect, disconnect. Then suddenly in the same session, the bottom bar showed PICO DISCONNECTED. Clicking caused an error to throw "command 'picogo.toggleConnect' not found". All other commands throw similar errors. Reset = "command 'picogo.reset.soft' not found", All Commands - "command 'picogo.listCommands' not found", Run - "command 'picogo.run' not found", Upload - "command 'picogo.uploadFile' not found".

## What were you expecting to happen?
 Saving the code.py file to restart the PI-PICO and run, as it was doing just a few moments before.

## Any logs, error output, etc?
*(If it’s long, please paste to https://gist.github.com and insert the link here)*
Mentioned above. Please recommend any additional log information needed.

## Any other comments?

## Support info

Copy this from the `Pico-Go > Help > Get support info` option in the command palette:

```
< Paste Here >
```

HopWorks, 2022-01-06:
Not sure what the "Support info" under "Any other comments?" is suppose to do. Clicking on "" throws a popup window...
POP UP ERROR WINDOW - Command 'Pick-Go > Help > Get support info' resulted in an error (command 'picogo.extra.getFullVersion' not found
[image]
)

HopWorks, 2022-01-06:
I open a terminal and get this... Could this be the cause?
------------
 Session contents restored from 1/6/2022 at 12:29:31 PM 

Exception in thread Thread-1:
Traceback (most recent call last):
  File "O:\Python\py390\lib\threading.py", line 950, in _bootstrap_inner
    self.run()
  File "O:\Python\py390\lib\threading.py", line 888, in run
    self._target(*self._args, **self._kwargs)
  File "c:\Users\Hop\.vscode\extensions\chriswood.pico-go-1.4.3\terminalExec.py", line 106, in runServer
    lsock.bind((ip, port))
OSError: [WinError 10048] Only one usage of each socket address (protocol/network address/port) is normally permitted

HopWorks, 2022-01-06:
Pico-Go still works on my laptop's VSCODE and it is version 1.63.0, where the version that is broken on my desktop is 1.63.2. Is there any way to roll back VSCODE or do I have to reinstall? Then do I have to set everything up again? What a nightmare.

HopWorks, 2022-01-06:
OK, so the PICO-GO extension works on my laptop and VS Code updated it's version to 1.63.2. So it has to be something else that is causing the issue. Again, strangely, I did not install or change anything. I paused to watch DVR'd football, and when I came back, I changed a couple lines of code in my CODE.PY, and when I tried to save it, my PICO was disconnected. When I went to reconnect, I got all the errors I mentioned here. And before my break I worked with my PICO with VSC for about 3 hours with no issues. NO IDEA what could be causing it. When I went to developer tools in VSC, I found ALL SORTS of red instances, mostly related to my setup with STM32. So I removed those extensions, but the problem persists. I also forgot to mention that I have PlatformIO and have had issues with it. When I start VSC, I see the Pico-Go bar at the bottom, and when I click CONNECT, I see PLATFORMIO loading pop up on the same bar. Maybe coincidence, maybe not. 
Anyway, I am going to rip VSC from my system and reinstall, and see what happens.

HopWorks, 2022-01-07:
Well I uninstalled and reinstalled VS Code and still the same issue. I tried to sync with my working laptop to my local setup and still doesn't work. I am out of options. I hope someone can come up with a fix for my issue.

[image]

HopWorks, 2022-01-07:
I am back up and running. I had to remove VSCODE completely, not just uninstall. There must have been a conflict with another extension. As I add those back in, I'll watch to see which one was responsible and post it here in case someone else experiences this. If I do not find it by tomorrow, I'll close this issue anyway. THANKS

haflinger, 2022-01-12:
I @HopWorks try to create a pico-go.json file at the root of your workspace with this content : 

```json
{
    "sync_folder": "",
    "open_on_start": true,
    "statusbar_buttons": ["status", "connect", "upload", "download", "run", "softreset", "global_settings", "project_settings"]
}
```

steeley, 2022-01-21:
I have exactly the same issue on VScode  1.63.2  on max osx. It worked once after I installed the extension. Now it doesn't work at all. tried the JSON as above made no difference.

haflinger, 2022-01-21:
You have the same error in the VSCode output ?

steeley, 2022-01-21:
Sorry I don't know what' VScode output' you mean - may the debug output...

I see these errors in the debug window from developer tools:
`(node:2312) [DEP0005] DeprecationWarning: Buffer() is deprecated due to security and usability issues. Please use the Buffer.alloc(), Buffer.allocUnsafe(), or Buffer.from() methods instead.(Use `Electron --trace-deprecation ...` to show where the warning was created)
I @ console.ts:137
`

and the message when you press a button on the blue control bar:

`notificationsAlerts.ts:42 command 'picogo.toggleConnect' not found
onDidChangeNotification @ notificationsAlerts.ts:42
`

stephenpcook, 2022-01-26:
I have the same issue - installed yesterday, was working fine, have disabled then reenabled it and all the buttons give `command 'picogo.*' not found`.

I disabled it globally and attempted to reenable it just for the workspace (because I didn't like the status buttons appearing on non-pico workspaces - perhaps that's a feature request).  Have since reenabled it globally, uninstalled, tried to install an older version and then update, tried completely deleting the `C:/users/me/.vscode/chriswood.pico-go-1.4.3` and reinstalling but no luck.

In output:log (Extension Host) there is the following line, no errors/warnings:
```
[2022-01-26 18:11:45.108] [exthost] [info] ExtensionService#_doActivateExtension ChrisWood.pico-go, startup: true, activationEvent: '*'
```

jsimonrichard, 2022-01-29:
I'm having a similar issue. None of the commands work (after, but for me I think the root cause is some sort of serial port issue. According to my vscode extension log file, the npm module `@serialport` is trying to require a library that doesn't exist on my machine: `GLIBC_2.28`. By the way, I'm running pop-os, which is a fork of ubuntu.

playertr, 2022-02-27:
I think I'm getting the same error. Here's the complete stack trace from `exthost.log`. I'm on Ubuntu 20.04 on my desktop.

<details> 
<summary> Stacktrace </summary>

```
ExtensionService#_doActivateExtension ChrisWood.pico-go, startup: false, activationEvent: '*'
[2022-02-27 08:18:05.417] [exthost] [error] Activating extension ChrisWood.pico-go failed due to an error:
[2022-02-27 08:18:05.417] [exthost] [error] Error: /snap/core/current/lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.28' not found (required by /home/tim/.vscode/extensions/chriswood.pico-go-1.4.3/node_modules/@serialport/bindings/lib/binding/node-v89-linux-x64/bindings.node)
	at process.func [as dlopen] (electron/js2c/asar_bundle.js:5:1846)
	at Object.Module._extensions..node (internal/modules/cjs/loader.js:1185:18)
	at Object.func [as .node] (electron/js2c/asar_bundle.js:5:1846)
	at Module.load (internal/modules/cjs/loader.js:982:32)
	at Module._load (internal/modules/cjs/loader.js:823:14)
	at Function.f._load (electron/js2c/asar_bundle.js:5:12913)
	at Function.i._load (/snap/code/88/usr/share/code/resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:112:32271)
	at Function.p._load (/snap/code/88/usr/share/code/resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:112:28855)
	at Function.v._load (/snap/code/88/usr/share/code/resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:107:60382)
	at Module.require (internal/modules/cjs/loader.js:1006:19)
	at h (/snap/code/88/usr/share/code/resources/app/out/vs/loader.js:4:699)
	at bindings (/home/tim/.vscode/extensions/chriswood.pico-go-1.4.3/node_modules/bindings/bindings.js:112:48)
	at Object.<anonymous> (/home/tim/.vscode/extensions/chriswood.pico-go-1.4.3/node_modules/@serialport/bindings/lib/linux.js:2:36)
	at Module.u._compile (/snap/code/88/usr/share/code/resources/app/out/vs/loader.js:4:1313)
	at Object.Module._extensions..js (internal/modules/cjs/loader.js:1155:10)
	at Module.load (internal/modules/cjs/loader.js:982:32)
	at Module._load (internal/modules/cjs/loader.js:823:14)
	at Function.f._load (electron/js2c/asar_bundle.js:5:12913)
	at Function.i._load (/snap/code/88/usr/share/code/resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:112:32271)
	at Function.p._load (/snap/code/88/usr/share/code/resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:112:28855)
	at Function.v._load (/snap/code/88/usr/share/code/resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:107:60382)
	at Module.require (internal/modules/cjs/loader.js:1006:19)
	at h (/snap/code/88/usr/share/code/resources/app/out/vs/loader.js:4:699)
	at Object.<anonymous> (/home/tim/.vscode/extensions/chriswood.pico-go-1.4.3/node_modules/@serialport/bindings/lib/index.js:14:22)
	at Module.u._compile (/snap/code/88/usr/share/code/resources/app/out/vs/loader.js:4:1313)
	at Object.Module._extensions..js (internal/modules/cjs/loader.js:1155:10)
	at Module.load (internal/modules/cjs/loader.js:982:32)
	at Module._load (internal/modules/cjs/loader.js:823:14)
	at Function.f._load (electron/js2c/asar_bundle.js:5:12913)
	at Function.i._load (/snap/code/88/usr/share/code/resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:112:32271)
	at Function.p._load (/snap/code/88/usr/share/code/resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:112:28855)
	at Function.v._load (/snap/code/88/usr/share/code/resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:107:60382)
	at Module.require (internal/modules/cjs/loader.js:1006:19)
	at h (/snap/code/88/usr/share/code/resources/app/out/vs/loader.js:4:699)
	at Object.<anonymous> (/home/tim/.vscode/extensions/chriswood.pico-go-1.4.3/node_modules/serialport/lib/index.js:2:17)
	at Module.u._compile (/snap/code/88/usr/share/code/resources/app/out/vs/loader.js:4:1313)
	at Object.Module._extensions..js (internal/modules/cjs/loader.js:1155:10)
	at Module.load (internal/modules/cjs/loader.js:982:32)
	at Module._load (internal/modules/cjs/loader.js:823:14)
	at Function.f._load (electron/js2c/asar_bundle.js:5:12913)
	at Function.i._load (/snap/code/88/usr/share/code/resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:112:32271)
	at Function.p._load (/snap/code/88/usr/share/code/resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:112:28855)
	at Function.v._load (/snap/code/88/usr/share/code/resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:107:60382)
	at Module.require (internal/modules/cjs/loader.js:1006:19)
	at h (/snap/code/88/usr/share/code/resources/app/out/vs/loader.js:4:699)
	at Object.<anonymous> (/home/tim/.vscode/extensions/chriswood.pico-go-1.4.3/lib/connections/pyserial.js:20:19)
	at Module.u._compile (/snap/code/88/usr/share/code/resources/app/out/vs/loader.js:4:1313)
	at Object.Module._extensions..js (internal/modules/cjs/loader.js:1155:10)
	at Module.load (internal/modules/cjs/loader.js:982:32)
	at Module._load (internal/modules/cjs/loader.js:823:14)
	at Function.f._load (electron/js2c/asar_bundle.js:5:12913)
	at Function.i._load (/snap/code/88/usr/share/code/resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:112:32271)
[... cut ...]
	at w._doActivateExtension (/snap/code/88/usr/share/code/resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:98:13341)
	at w._activateExtension (/snap/code/88/usr/share/code/resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:98:12293)
	at processTicksAndRejections (internal/process/task_queues.js:93:5)
	at async Promise.all (index 0)
	at async w.$activate (/snap/code/88/usr/share/code/resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:98:21810)
```

</details>

playertr, 2022-02-27:
Update: it appears the bug involved the installation of vscode via snap, which somehow interacted with npm's version of serialport. Following the instructions in this related [issue](https://github.com/cpwood/Pico-Go/issues/96#issuecomment-962195465) to reinstall Code from a .deb file fixed it.

roaldarbol, 2022-03-31:
Also doesn't work for me on MacOS with `VS Code Insiders` installed with `brew`, however using the normal `VS Code` installed the same way works...

arnytrty, 2022-03-31:
I dont think this is bug related to snap. Yesterday 2022.03.30 **VSCode** on my **Windows** machine updated and the extension stopped working, as described above (_no commands found, no buttons showing_). I installed **VSCodium** on my **Debian** machine and installed the extension via vsix file, and it didnt work either (_this was working before too_).

Erriez, 2022-04-30:
I've installed Pico-Go v1.4.3 in Visual Studio Code v1.66.2, installed official `code_1.66.2-1649664567_amd64.deb` on Ubuntu 21.10 and can't configure the project via `CTRL+SHIFT+P  Pico-Go -> Configure Project`:

Error: `Command 'Pico-Go > Configure project' results in an error (command 'picogo.initialise' not found)`.

[image]

[image]

Any suggestions how to proceed?