cpwood/Pico-Go #115: when i use the arrow up key to retake a command i got un python exception and the terminal is frezzing

dapiced, 2022-01-10:
>>> Exception in thread Thread-2 (listenForInput):
Traceback (most recent call last):
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.10_3.10.496.0_x64__qbz5n2kfra8p0\lib\threading.py", line 1009, in _bootstrap_inner
    self.run()
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.10_3.10.496.0_x64__qbz5n2kfra8p0\lib\threading.py", line 946, in run
    self._target(*self._args, **self._kwargs)
  File "c:\Users\Dom-Samsung\.vscode\extensions\chriswood.pico-go-1.4.3\terminalExec.py", line 128, in listenForInput
    for ch in getCharacterWindows():
  File "c:\Users\Dom-Samsung\.vscode\extensions\chriswood.pico-go-1.4.3\terminalExec.py", line 78, in getCharacterWindows
    c = msvcrt.getch().decode("utf-8")
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xe0 in position 0: unexpected end of data

thanks
Dominic
[image]

dapiced, 2022-01-11:
HI, as a temporary workaround to prevent the **frezzing**..., i have change this in the file : c:\Users\Dom-Samsung.vscode\extensions\chriswood.pico-go-1.4.3\**terminalExec.py**", **line 78, in getCharacterWindows**   **LIKE THIS :**

-------------------------------------------------------------------------
def getCharacterWindows():
    while True:
        c = msvcrt.getch().decode**("utf-8")**
        yield c     
        time.sleep(characterDelay)
-------------------------------------------------------------------------

**WITH**

-------------------------------------------------------------------------
def getCharacterWindows():
    while True:
        c = msvcrt.getch().decode(**"latin-1")**
        yield c     
        time.sleep(characterDelay)
------------------------------------------------------------------------

BUT, it appear to be a bug, as least what i understand...thanks

Dominic

M-Kub, 2022-01-23:
I have the exact same issue. Didn't test the workaround yet, I first have to find the time to look into it.