cpwood/Pico-Go #123: Visual Studio Code 1.66.2: "Command 'Pico-Go > Connect' resulted in an error (command 'picogo.connect' not found)"

TerrificTable, 2022-04-30:
## What are the steps to reproduce this issue?
(0. Connect Raspberry Pi Pico to PC)
1. Install Pico-Go extension
2. (The click menu thingys on the bottom won't show up)
3. ctrl+p -> >Pico-Go: Connect

## What happens?
Buttons on the bottom of VS Code won't show up,
CTRL+P: '>Pico-Go: Connect': Error window opens and shows this message `Command 'Pico-Go > Connect' resulted in an error (command 'picogo.connect' not found)`

## What were you expecting to happen?
Pico-Go connecting to my Pico,
A Terminal to open and being able to execute (Python) code in it

## Any logs, error output, etc?
No, it seams like Pico-Go doesn't load compleatly

## Any other comments?
Same thing happens to every other Pico-Go command just with for example: Command 'Pico-Go > Configure  #project' ... 'picogo.initialise' not found

## Support info

Copy this from the `Pico-Go > Help > Get support info` option in the command palette:
Returns same error as before just with: `Command 'Pico-Go > Help > Get support info' ... 'picogo.extra.getFullVersion' not found`

TerrificTable, 2022-04-30:
Oh and Autocompletion doesn't work

kraemrz, 2022-05-02:
same problem here :(

RCPhysics, 2022-05-02:
I believe this is a duplicate of #118.  For now you can use an older version of VSCode, and it should work.

slowpokefarm, 2022-05-05:
It seems that Pico-Go has been finally abandoned. This issue exists for some time now, there are many reports of it and even Pull Request to fix it but it remains unmaintained. Also the website with documentation is now down.

tswallen, 2022-05-07:
Same issue here

postfen, 2022-05-07:
Really hoping this gets updated, it's the best extension I've found for vscode development with the pico.

Idirianuk, 2022-05-07:
> Really hoping this gets updated, it's the best extension I've found for vscode development with the pico.

It's worth pointing out that the forked and fixed repo available here...

https://github.com/NixM0nk3y/Pico-Go/tree/develop

...builds into a version which, given my limited testing, seems to work satisfactorily. There are instructions present on how to build and install it for yourself. It's not overly difficult. Just install a recent version of node.js and npm and go from there.

TerrificTable, 2022-05-08:
Got a Question, is it posible tu Use PlatformIO to do the Same thing PicoGO die?

RCPhysics, 2022-05-12:
> It's worth pointing out that the forked and fixed repo available here...
>
> https://github.com/NixM0nk3y/Pico-Go/tree/develop
>
> ...builds into a version which, given my limited testing, seems to work [...]

I've tried this on MacOS 12.3.1, and no luck for me.  I get one error during compile, but I'm not sure if it's relevant or not, but I'm posting it here in case it helps anyone.  I don't have enough javascript experience to diagnose.

`npm WARN EBADENGINE Unsupported engine {
npm WARN EBADENGINE   package: 'promirepl@2.0.1',
npm WARN EBADENGINE   required: { node: '^8' },
npm WARN EBADENGINE   current: { node: 'v17.8.0', npm: '8.10.0' }
npm WARN EBADENGINE }`

postfen, 2022-05-12:
> It's worth pointing out that the forked and fixed repo available here...
>
> https://github.com/NixM0nk3y/Pico-Go/tree/develop
>
> ...builds into a version which, given my limited testing, seems to work [...]

Thank you for pointing this out, this version works great on my desktop running Pop OS 22.04 and VSCode 1.67.1. 

However, like @RCPhysics, it doesn't agree with my M1 MacBook running MacOS 12.3.1, which is unfortunate.

Idirianuk, 2022-05-12:
> Thank you for pointing this out, this version works great on my desktop running Pop OS 22.04 and VSCode 1.67.1.
>
> However, like @RCPhysics, it doesn't agree with my M1 MacBook running MacOS 12.3.1, [...]

The .VSIX file you have built, AIUI, should be usable on vs.code running on any computer. :)

postfen, 2022-05-12:
> The .VSIX file you have built, AIUI, should be usable on vs.code running on any computer. :)

I was hoping so, but it still gives me the "Command Pico-Go > ____" only on the Mac.

LioBou, 2022-05-16:
> It's worth pointing out that the forked and fixed repo available here...
>
> https://github.com/NixM0nk3y/Pico-Go/tree/develop
>
> ...builds into a version which, given my limited testing, seems to work [...]

Hi, I could not find the instructions to build and install ? Could you precise ? (I'm running Linux Mint 20.3)
Thanks!

postfen, 2022-05-16:
> Hi, I could not find the instructions to build and install ? Could you precise ? (I'm running Linux Mint 20.3) Thanks!

Hello, I used this; https://github.com/cpwood/Pico-Go/blob/main/development.md
It is located in the main branch.

PettechLabs, 2022-06-07:
Having the same issue. Any fix out there yet?

wzaremba, 2022-06-12:
I tried to build but I think I have something wrong with my env. The build got stuck on this step:
src/stubs/stubs-manager.js -> lib/stubs/stubs-manager.js
No idea - as I am poor developer - where to look for help. 
npm is updated. Node looks updated. 
Thank you in advance.

Idirianuk, 2022-06-12:
> I tried to build but I think I have something wrong with my env. The build got stuck on this step: src/stubs/stubs-manager.js -> lib/stubs/stubs-manager.js No idea - as I am poor developer - where to look for help. npm is updated. Node looks updated. Thank you in advance.

It's okay at that point. I assume you're building it under Windows. Mine does the same. Just close the window that seems to be stuck after a minute (to make sure it's finised) and just carry on.

I've just tried this myself on Windows (I normally use Linux) and CAN verify that it does all work and I've just created and installed my own build of the .vsix.

The only error I did have was during the `npm install` phase where I had to run the command `npm install node-gyp`

Hope this gets you moving again. :)

wzaremba, 2022-06-12:
Thank you. I am using Mac, and I was fighting with all kind of
dependencies. I think but I overcome this and I was able to generate the
package, but it was not not working :).
Actually with this extension, I receive strange behaviour. The tool look
connected, but actually I cannot download and upload anything. And I cannot
run anything. I will have to spend more time on debug - I know that the
board is reacting, as via Thonny all works.
Thank you all for help.

W.

------------------------------
*Od:* Idirian ***@***.***>
*Wysłane:* Sunday, June 12, 2022 5:54:17 PM
*Do:* cpwood/Pico-Go ***@***.***>
*DW:* Wojciech Zaremba ***@***.***>; Comment <
***@***.***>
*Temat:* Re: [cpwood/Pico-Go] Visual Studio Code 1.66.2: "Command 'Pico-Go

> Connect' resulted in an error (command 'picogo.connect' not found)"
(Issue #123)

I tried to build but I think I have something wrong with my env. The build
got stuck on this step: src/stubs/stubs-manager.js ->
lib/stubs/stubs-manager.js No idea - as I am poor developer - where to look
for help. npm is updated. Node looks updated. Thank you in advance.

It's okay at that point. I assume you're building it under Windows. Mine
does the same. Just close the window that seems to be stuck after a minute
(to make sure it's finised) and just carry on.

I've just tried this myself on Windows (I normally use Linux) and CAN
verify that it does all work and I've just created and installed my own
build of the .vsix.

The only error I did have was during the npm install phase where I had to
run the command npm install node-gyp

Hope this gets you moving again. :)

—
<https://github.com/cpwood/Pico-Go/issues/123#issuecomment-1153214578>, or
unsubscribe

kraemrz, 2022-06-17:
> I have built and packaged the version here https://github.com/NixM0nk3y/Pico-Go/tree/develop
> 
> and hosting it at the following address: https://jasongi.com/static/pico-go-1.4.4.vsix
> 
> I will take it down once @cpwood publishes the fix to the vscode extension marketplace
> 
> Works fine for me now on my Intel Mac with latest VSCode
> 
> Download the file and install with
> 
> ```
> code --install-extension pico-go-1.4.4.vsix
> ```
> 
> or via the menu in the extensions panel
> 
> [image]
> 
> Hopefully this helps with people struggling to get this working.

Thank you jasongi!!!
Finally I can use this extension again :)