cpwood/Pico-Go #125: Cannot Initialize Project

l-gothberg, 2022-06-25:
## What are the steps to reproduce this issue?
Hardware Configuration:
 -- M1 Mac Mini (Apple Silicon) running macOS 12.4 (21F79)

1. Install latest MicroPython UF2 on new Pi Pico board -- (currently 20220618_v1.19.1.uf2 )
2. Install VS Code
3. Install Python 3
4. Install Pico-Go extension in VS Code
5. Open empty folder for new project
6. Open command palette (Cmd+Shift+P), select "Pico-Go: Configure Project"

## What happens?
Receive the following error:
`Command 'Pico-Go > Configure project' resulted in an error (command 'picogo.initialise' not found)`

## What were you expecting to happen?
Unsure.  Probably a board selection menu?  I have not used the Pico-Go extension before.

## Any logs, error output, etc?
*(If it’s long, please paste to https://gist.github.com and insert the link here)*
No logs, no error output not specified elsewhere.

## Any other comments?
Seems that selecting any of the Pico-Go options from the command palette produces similar errors, with slightly different file names related to the command chosen.

## Support info

Copy this from the `Pico-Go > Help > Get support info` option in the command palette:

None.  Receive similar error message to the one noted above.
```
Command 'Pico-Go > Help > Get support info' resulted in an error (command 'picogo.extra.getFullVersion' not found)
```

steeley, 2022-06-29:
Same issue too....

this software is dead so don't waste your time. it should be taken off the VS Marketplace

cpwood, 2022-06-29:
Agreed. I've just pulled it and am closing the project.

closed 2022-06-29 by cpwood